Creality3DPrinting/Ender-3 #1: please re upload solidworks ender 3 part file

quocanhcgd, 2018-07-03:
please re upload solidworks ender 3 part file. All it error. I think you don't change name from chinese to english.Thanks

flowerrobot, 2018-07-03:
Can confirm SW2017 SP5 cannot open these files. I suspect more of a corruption than a language issue.

sexycyborg, 2018-07-04:
Will try, please be patient working on a lot of issues simultaneously.

quocanhcgd, 2018-07-04:
I try with 2015, 2016, 2018 but not luck. Can't open anything (part, assem..)

saLLadEX, 2018-07-04:
I can't open too, i tried to open on SW2017 SP1.

Yeziwaldens, 2018-07-07:
没有一个能打开的  我用stp存储了零件然后覆盖 还是出错，N多的坏面

pyromanci, 2018-07-17:
I have been able to open the Assembly files. But with the eDrawing Viewer, not solid works. 

Did notice something intereing int he Ender3-1.sldasm file:

[image]

interesting to see this one.

TLWhite, 2018-07-24:
I understand all the discussion about the primary purpose repository and version tracking.  The software coding guys are absolutely right about it from their point of view.  
  With that said, those of us on the mechanical side that would like to import a workable model into CAD program, would it be possible for someone at Creality to use the "Pack and Go" function of the CAD program, then upload the generated Pack and Go file.  Yes, it is called a ZIP, and it will have version tracking issues for die-hard git hub users.  However, it will give us an assembly that we can work with-without converting over 400 part files and then assembling them.  

I really don't want to get in the middle of that other conversation on file tracking and the purpose of Git.  I just want a working CAD Assembly File (As I'm guessing a few others do also).

Thank you,

sexycyborg, 2018-07-24:
Completly reasonable, let me get a bunch of stuff, are you folks willing to help me sort out what's useful so I can update things?

TLWhite, 2018-07-25:
Sure, I’d be glad to help.  The beauty of that “Pack and Go” within Solidworks, is that it includes all the linked files that are needed, and not the ones that aren’t.  Plus a lot more benefits relating to the assembly package.  (Pardon me if you already know all that).  

Just talk to the CAD person at Creality, he will know what we are talking about.  He may also have some recommendations on how we track mechanical change-logs to compensate for the revision issues outside of the normal github.

techmach, 2018-07-26:
I have one.  Who wants it?  As long as you are running IronCAD you can have it.

techmach, 2018-07-26:
[image]

quocanhcgd, 2018-07-27:
Techmach please share it. Thanks. My email: quocanhcgd@gmail.com.

sexycyborg, 2018-08-01:
I'm having a hell of a time getting this stuff- I'll keep asking and try to find an angle. They're a little disenchanted with OSH at the moment. There's no excuse, I know. I'm really sorry about this and will keep trying😔

shoreliner, 2018-08-02:
You should try and disenchant them as much as possible... major US printing companies are going open source/open hardware.  I will be in Shenzhen in the next few months to start a project, perhaps we could put a discussion group together.

techmach, 2018-08-02:
Thanks for responding.  From what I've seen especially with folks like Lulzbot there are several degrees of OSH.  I've been a TAZ 3 user for 4+ years and their idea is a mix of Openscad and FCad files interspersed with SW files.  I'm sure they did their designs in SW but they still have a mix of CAD formats on their site which makes no sense but that's how they roll.   

If there is a fear of cross pollination with other companies I wouldn't sweat that because at the commercial level there are few secrets in China as you know.  These 3D printers aren't like full blown CNC metal cutting machines so maybe someone there at Creality will realize that posting the SW files isn't that big of a deal.  The STEP files were enough to build an assembled model from.  The important bits are there.

rulas9327, 2018-08-03:
I have a fatal solidworks error, i cant use the solidworks filesin 2018 sp1 please re upload and thx naomi for your awesome work

acristallini, 2018-08-13:
Hi,
which version was used to save this project ? 
All files are reported as "saved with a future version" of SW.
I use SW 2018 SP1
Thanks in advance

SIeeepy, 2018-08-17:
Hi,
I also faced errors when trying to open the published Ender 3 Solidworks files, so I've used the provided STEP files to create a Solidworks 2015 assembly.
[image]
Some models were missing and I had to create them. I've used the "pack and go" feature. 
You can **download the CAD files** here (~85mb): https://s3-eu-west-1.amazonaws.com/krimskrams/Ender_3_CAD_Solidworks2015_STEP_IGS.zip
Updated STEP and IGS files are included. Hope this is ok with you @sexycyborg .

sexycyborg, 2018-08-17:
This is great thank you! I'll put in the repo shortly.

Once again sorry for the delays everyone. Juggling a lot of things and the engineers are all using old pirated versions of everything I think. Thanks so much for your patience...

Agniusm, 2018-08-17:
Perfect, thanks

croplaninec, 2018-09-03:
thank you sleeepy :-)

steenerson, 2019-12-13:
Looks like this is resolved, closing out this old issue.

closed 2019-12-13 by steenerson