Creality3DPrinting/Ender-3 #2: PCB Design

alekssobolewski, 2018-07-03:
Hi Naomi!

Can you please upload the PCB files in format that does not require a paid PCB design software?
Or at least Eagle files.

What program creates that .PCB file?

Moorviper, 2018-07-03:
try kicad i think this could open it

NikiSchlifke, 2018-07-03:
>try kicad i think this could open it

Probably not.
A pcb layout isn't sufficient anyway, for open hardware we need the circuit diagram, the part libraries **and** the layout.

alekssobolewski, 2018-07-03:
Exactly @NikiSchlifke.
I already tried some programs, and it turned out to be an Altium Designer PCB layout. There is no elements and schematic within the file, so it is basically useless :/

NikiSchlifke, 2018-07-03:
>I already tried some programs, and it turned out to be an Altium Designer PCB layout.

Well it's the status quo in industry... as for us Hardware Hackers we really need KiCad

@Creality3DPrinting 
@sexycyborg 
Have a look at this, maybe this will help us out:
https://github.com/thesourcerer8/altium2kicad

Moorviper, 2018-07-03:
ok it can't open it but it would be nice.
Prusa also switched to kicad

sexycyborg, 2018-07-04:
I agree KiCAD would be best, I'll see if something can be done.

RudolphRiedel, 2018-07-04:
I am missing the schematics as well.
And the Ender-3.PCB is not really an Altium Designer file.
Altium uses .PcbDoc for the layout files.
This is an old Protel format that AD happens to be able to open, really old,
the last Protel version was 99SE. Altium Designer started in 2004.
It barely opens with two warnings generated for AD release 12 and 13.
The only component that has a 3D-model is the connector EXT-A2.

And there are really quite a number of real issues with this design.
But I have no idea if this repository means that we should point these out to Creality in order to let them eventually fix these.

alekssobolewski, 2018-07-04:
> And the Ender-3.PCB is not really an Altium Designer file.

@RudolphRiedel Well, maybe it is a Protel file. I just found it opens without a problem in Altium. I'm not a specialist in in, just downloaded a free demo to open a file.
But it does not matter, we need a real schematic and pcb design file created in a free software (like KiCAD).
[image]

RudolphRiedel, 2018-07-04:
> But it does not matter, we need a real schematic and pcb design file created in a free software (like KiCAD).

While yes, the schematic is missing I do not agree that Creality needs to supply the data in any form they are not actively using. In fact they do not "need" to supply anything and we can considers us lucky already that they supplied as much as they have now.
If asked nicely they may supply a set of .pdf files along with their original design data.
Or someone could just fork this repository and add more files.

And for what purpose do you really "need" the files?
Do you plan to build your own improved PCB? How many out there would do that?
I could do that and already with the data available but the important thing for me here is repairability rather than the need to create a new board from this.
Maybe tweak a few things.
If I wanted a new board I could chose one of the other available options like an Einsy-Rambo which btw. is released as Altium-Designer project.

RudolphRiedel, 2018-07-04:
Here, I just created a project for the board and created an assembly plan with my default output job and not editing anything.

[Ender3_pcb_parts.PDF](https://github.com/Creality3DPrinting/Ender-3/files/2164340/Ender3_pcb_parts.PDF)

alekssobolewski, 2018-07-05:
> And for what purpose do you really "need" the files?
Do you plan to build your own improved PCB? How many out there would do that?

Well, my CR-10 is currently connected to MKS L just because I like to do everything on my own, but I have also two not working Creality boards that I broken accidentally. It would be nice to have a schematic to repair the board or add something new. For example improve the 5V supply and maybe change Allegro steppers to Trinamic.

RudolphRiedel, 2018-07-05:
Repairability, exactly.
In the meantime this could be helpfull: https://github.com/RudolphRiedel/CR-10_wiring

I started to draw a schematic for the CR-10 controller-box a while ago, the whole thing including the pcb.
This is as complete as I needed it to be so far.
And the Ender-3 has the same controller-board as the CR-10, well at least later models came with the V1.1.2 version of the board we see in the Ender-3.PCB now.

With the PCB data I could easily improve my schematic further but at the same time I would prefer if Creality supplies the original schematics.

alekssobolewski, 2018-07-05:
Wow! Thanks a lot @RudolphRiedel 😃 I need to replace the stepper drivers, so now I can make an adapter to plug them for example on breadboard (I really have to learn how to properly solder QFN uCs 😛).

RudolphRiedel, 2018-07-05:
It is a start, I was more interested in the controller. And with the wiring of the box I tried to figure out a way to add an auto-bed-level sensor more elegantly to the control-box.

Soldering QFNs is not really that difficult. An affordable hot-air station, flux, something to mask off the rest of the nearby area of the board like capton-tape.

The A4988 only have a QFN-28 case and the board has only two layers, so these should come of quite easily.

Regarding tweaking, I was thinking about changing the config for the Z-Stepper from 1/16 steps to full steps.
And now that we have the layout I can check that in detail without the need to remove the heatsinks.
It turns out that each stepper driver has resistors on every single config line including MS1, MS2 and MS3 which control the microstepping resolution.
I could easily remove R35, R36 and R37 and connect their pins that lead to U7 with GND.
Sure that reduces the resolution of the Z-axis to 0.04 mm per step and you have to keep that in mind with the layer-heights. But going full step gives it more strength.

Or in other words, Creality delivered, if not all than most of it and in some regards with a suprising level of detail. We can hope that they are so kind to provide even more information but it is up to the community now to take part in this.
Stop demanding that everything is handed to you, apreciate what you got and do something with it!

And no, I am not adressing you specifically with this Aleks, more the community overall. :-)

I did not like the supplied BOM, so I exported a new one and started editing it.
[Ender-3_BOM.xlsx](https://github.com/Creality3DPrinting/Ender-3/files/2167733/Ender-3_BOM.xlsx)
Miss something? Add it!

digitaljunk, 2018-07-06:
@RudolphRiedel thanks for your efforts. Great work.
But Creality should fulfill the OSHW certification. Especially when they start to do marketing with that OSHW and Open Source hardware term.

@Creality3DPrinting please publish all files in a file format that can be used for improving the hardware.

RudolphRiedel, 2018-07-06:
@digitaljunk 

The schematic is missing, okay. And probably the project file and a few files around them.
Above the parts library has been requested but the parts are just embedded into the PCB file and can be generated from it: [Ender-3_PcbLib.zip](https://github.com/Creality3DPrinting/Ender-3/files/2171505/Ender-3_PcbLib.zip)

But otherwise releasing the original design files in whatever format is fully compliant with the OSHW definition: https://www.oshwa.org/definition/

For crying out load, they went this far and released their original data.

If it is not in some weird custom format no one ever has heard of you will find someone who can convert it. Regarding converting, I took a quick look at the https://github.com/thesourcerer8/altium2kicad from above and it uses .PcbDoc files from Altium Designer and not .PCB files.
So I just saved the file with AD17: 
[Ender-3_PcbDoc.zip](https://github.com/Creality3DPrinting/Ender-3/files/2171524/Ender-3_PcbDoc.zip)

Have fun with it and post a KiCad file if you can.

Would you like a set of Gerber files as well?

sexycyborg, 2018-07-07:
Guys, it's literally just me doing this and there's a million other things going on. Unless your 3D printing parts for your failing iron lung cut me a little slack and give it some time😉

DuncanAmos, 2018-08-11:
Let's get things in perspective:

Creality did their design in what is probably the best Schematic/PCB/Workflow/3D combined package out there - why the heck wouldn't they?

Eagle is a far more common 'maker/enthusiast/hobbyist' package that KiCAD so, for a moment, let's get real...

Altium does have the capability of exporting their design files in file form called ACCEL ASCII format. This is a data structure EAGLE is able to import. If you wish to import a schematic, click on File/Import/Pcad, Altium, protel option. Then follow the prompts on the dialog box.

I don't know what file imports exist in KiCAD, but maybe this is a start?

Naomi is doing her best - thank you Naoni...

sexycyborg, 2018-08-15:
I just uploaded a bunch of PCB files that I got in different formats, sorry I'm a bit out of my depth if they are useful, or can be converted to something useful I can also upload please let me know.

https://github.com/Creality3DPrinting/Ender-3/tree/master/Ender-3%20PCB

UrsBruetsch, 2018-08-15:
I'm a happy user of Altium and my idea is to change the Atmel 8bit processor to ESP32 for more power of the printer and maybe (later) for network compatibility. But it is exhausting and time-consuming to develop the circuit diagram from the board. So it would be great if the documents were complete. So that the community could participate in the development of Ender, CR10 and other...

DuncanAmos, 2018-08-15:
Urs I don't think anyone would disagree with you that schematics and board
files for all the popular packages would be ideal...

Natalie is doing her best but, as she freely admits, she's out of her
depth, so it's up to us to step up and make the process easier.

What's the first thing we need to know? Which software package are Creality
using to generate the schematic and board files - right? That's something
that Natalie can probably discover fairly easily.

Once we know that, then Natalie may be able to obtain 'latestVersion.abc'
and 'latestVersion.xyz' files, from which we can (hopefully) create
conversions for other software packages and provide them here, with
supporting information.

A specific request is likely to get a more positive and helpful response.

On a slightly different angle - why would you want to swap out the Atmel 8
bit processor for an ESP32?

Octoprint does a fantastic job of networking the printer already (and is a
'de facto standard') and is there really a requirement for 'more power' in
a device that is, basically, driving stepper motors and
monitoring/controlling two temperatures? All that is well within the
capabilities of the existing hardware.

I can see a need to split out the two fans, but beyond that, the existing
hardware works and, in general, works very well - even if it is close to
using all the available memory.

I'm a happy user of Altium and my idea is to change the Atmel 8bit
processor to ESP32 for more power of the printer and maybe (later) for
network compatibility. But it is exhausting and time-consuming to develop
the circuit diagram from the board. So it would be great if the documents
were complete. So that the community could participate in the development
of Ender, CR10 and other...

UrsBruetsch, 2018-08-15:
Duncan, thanks for your answer. Yes, you're right, it's not really necessary to change the Atmel to ESP.
And you can go with RasPi and Octoprint. That's a safe solution and it works
but it is also a complicated way to print. 
Wifi is defacto a standard in the printer industry.
The ESP also has the advantage that it is much faster than an 8-bit Atmel. 

Solutions like Klipper are very charming, but you always have to make the detour via a RasPi and reprogram the printer hardware. Not very comfortable for the people (beginners).
My idea is to update the printer electronics directly to have more speed and comfort without special external hardware and software. The RasPi solution costs more time and money to install and requires a lot of expert knowledge ...
So to bring the printer into a network and improve the speed, the ESP is certainly a good solution.

RudolphRiedel, 2018-08-15:
Sorry Naomi, but I went thru the files and do not see how these would be of any more use than the ones we already had,

The *schematic* is missing and not more versions of the same board file.

And btw. the board file has no 3D information except for one connector that is not even populated for the Ender-3 but replaced with a capacitor.
Plus the PCB has a few pretty obvious design flaws.
Without schematic and project file and everything else connected to the pcb-project it is however not that easy to contribute anything.

Would Creality even be interested in contributions?

RudolphRiedel, 2018-08-15:
I am pretty sure that Naomi knows what a schematic file is even when not beeing told to lock for a file called .sch or .schdoc.

RudyFiero, 2018-08-15:
My two cents. Do you really think that Naomi can just go looking for whatever engineering files she can find, that she has access to the engineering documents they have? I would think not.  

If she isn't getting the kind of material that would be useful then I think it is because whoever she is dealing with is either not being as helpful, or they have been directed to not provide very useful information.

DuncanAmos, 2018-08-15:
I wasn't implying in any way that Naomi wouldn't know a schematic file,
merely saying that the process of her obtaining the files for those who
seem to disparage her efforts so far, might be easier if we actually knew
what software was in use, and thus which file formats are Involved.

The schematic may be a ".sch" but it's very possible that it's not...

I certainly agree with the first part of Rudy's post and I think that the
second part is a distinct possibility.

Although Creality are supposedly open-sourcing things, that may not be 'all
at once' and it certainly won't be in file formats to suit everyone...

sexycyborg, 2018-08-15:
Okay so are we saying the new files I just uploaded are not useful? What format that can be exported from Altium Designer would be useful?

No, I don't know a whole lot about PCBs, no insult there or anything. That's why I'm asking.

RudyFiero, 2018-08-16:
I use a layout program from Mentor Graphics called PADS. It is supposed to open Altium Designer files. The Ender-3.PcbDoc has the right extension and maybe I can open that. Unfortunately I will not be able to try it until tomorrow. I need to use a dongle (hardware key)  to run the program and it is at work. 

So if I can translate the pcb file I would be able to move things (traces, components) on the layout but I couldn't do anything else with it. I couldn't add or remove parts or connections. For this we would also need the schematic files. I expect it would be called Ender-3.SchDoc. 

I have never imported Altium Designer files. I just looked at a couple of Youtube videos on importing them to PADS. It seemed straightforward but I will need to try it.  

@UrsBruetsch  You said you are a happy user of Altium. Were you able to open any of the pcb files.

There is an online Altium to KiCad converter. http://www2.futureware.at/KiCad/ 
https://github.com/thesourcerer8/altium2kicad
Since I don't have KiCad I didn't try it.

RudolphRiedel, 2018-08-16:
>Okay so are we saying the new files I just uploaded are not useful?

No, we are not saying that the files are not usefull, they might be for someone. :-)
I wrote more usefull than the first file because the new ones do not add any depth for me.

>What format that can be exported from Altium Designer would be useful?

Asked like this nothing else.
I already provided a file converted to AD17 to be used for the KiCad Converter.

But it would be nice to know what software Creality is really using, and exactly down to the version.
If that first Ender-3.PCB really is the original design file that Creality is using, they are not even using Altium Designer, they are using its predecessor which was Protel.

And a PCB project always has at least two files, the schematic and the pcb.
Naomi, do you have access to any more files?
In doubt just nake a screenshot and we help point out a few more files.

RudyFiero, 2018-08-16:
I am able to open and translate the **Ender-3.PcbDoc** file (only did one so far) for my PADS program. Some routing information has been lost (eg. ground plane) but that doesn't surprise me. I will try and have a better look at it tonight. 

At minimum the  **Ender-3.SchDoc** file (I assume that is what it is called) is what is needed.

RudolphRiedel, 2018-08-17:
There are revision 1.1.3 boards out?
https://github.com/Creality3DPrinting/Ender-3/issues/12

Can we please have design files for this one as well?
But regarding repairability and seeing that CR-10 printers are also built with the 1.1.2 boards I would very much apreciate this as additional set of files and have the 1.1.2 file-set also completed.

UrsBruetsch, 2018-08-17:
@RudyFiero Yes, i can open the "Ender-3.PcbDoc" but the Design Rule Check has more then 500 Errors.
I think, with some Work on the PCB-Rules and creating new Nets from connected Copper, it will be a usable PCB-File. 
[image]

RudolphRiedel, 2018-08-17:
Yes, I can confirm that, I just ran the DRC with AD17 and increased the amount of  violations reported to 5000.
Came out with 8 warnings and 629 rule violations.

I would guess though that a good portion of that are more or less false positives due to the file beeing converted from an older version.

But this like this are just silly:
[image]

The connection is not missing, the signal at pin 5 is named incorrectly, it should be SD_SCK or similar.

UrsBruetsch, 2018-08-17:
@RudolphRiedel Jepp, thats an Error (in Altium). 
Let's hope there are not too many of these mistakes on the board ...

RudolphRiedel, 2018-08-17:
I would not call that an error in Altium, that one is an error in the data.

I actually wonder now if Creality has a schematic for that board.

RudolphRiedel, 2018-08-20:
A friend asked me something about the FETs on board and so I had a closer look.
First is that these are named REP30N06LE in the board and the BOM while it is likely that these are RFP30N06LE.

Okay, we have an external FET for the bed and the fan does not draw much power which leaves the hot-end FET sending some amperes thru the board.
[image]
[image]

Looking at the GND from the board though I find no solid connection between the three FETs and the GND connector on the board from the PSU.
The same applies to the stepper drivers.

adamtegen, 2018-08-21:
I opened the files in a trial version of Altium and exported the top and bottom layers as pngs.  I made a pull request for them: https://github.com/Creality3DPrinting/Ender-3/pull/14

Additionally, I've gone through the exercise of tracing the lines for the 6-pin ISP connector and the 10-pin LCD connector.  I don't exactly know how/where is the best place to document those.  If anyone has any suggestions, please let me know.

Gazza-USA, 2018-09-15:
Guys, if you point me to the files I will take a look too.  I professionally use Altium and can take a look at the netlist etc.  If the native file is in another format (PADS, Allegro) send me that too.   If anyone has the schematics in any form too that would help.

Gazza-USA, 2018-09-15:
> A friend asked me something about the FETs on board and so I had a closer look.
> First is that these are named REP30N06LE in the board and the BOM while it is likely that these are RFP30N06LE.
>
> Okay, [...]

Yeah, I found exactly the errors you did.  They obviously don’t know how to use the tools properly as there are stacks of pads and component pads that are incorrectly defined.  Via stitching would solve the GND break and sorting out the polygon pour order would fix other things.  I did a cleanup myself for fun.  It looks like some things have been added after the schematics too like a resistor for an LED.  This is all just physical copper with no nets so it generates DRCs.

RudolphRiedel, 2018-09-15:
Have you even read the thread?
It's been 2 1/2 months now and there still is no schematic for the control-board from Creality.
It is like they do not even have a schematic.
Maybe this is annother contract job like they told us the modification of the software was.
And that the original .PCB is in a 20 year old Protel format without almost not 3D models is rather strange, together with  apprentice-level flaws like the grouping of C41/C42/C43.
I am tempted to re-work the  board but so far there has been no sign of apreciation of what is going on here whatsoever from Creality, so far this is a dead repository.

sexycyborg, 2018-09-15:
Sorry I've been working on getting the CAD files uploaded. What can I get you from them?

RudolphRiedel, 2018-09-15:
>What can I get you from them?

Well, the whole project directory they are actually using for development would be nice.
And as the same PCB is used at least in the CR-10 that is rather unlikely to have "Ender-3" in any of the files.
I even doubt that the actual board file is named xxx.PCB since that would mean the board is beeing developed with software that is 20 years old.
So, the real files would be a kicker, board.PcbDoc, board.SchDoc and board.PrjPcb at the very least if the board is indeed designed with Altium Designer. Plus board.PrjPcbStructure and the xxx.PcbLib, xxx.SchLib.

If that is for some reason not possible at the very least a Ender-3.SCH would be nice and of course the latest version of the Ender-3.PCB as this one was outdated even before it was released here.

sexycyborg, 2018-09-15:
Does anyone want to ask for something without calling the people being asked liars or making accusations of duplicity in the same post?

DuncanAmos, 2018-09-15:
Well said Naomi...!!!

RudyFiero, 2018-09-15:
Could we please get the current documentation for board. The Ender 3 I have has a pcb revision of 1.1.3. 

Personally I want the schematic. I have no intention of ever producing a replacement board. I want accurate information so that I make proper decisions on firmware upgrades.

sexycyborg, 2018-09-15:
Any documentation will be in Chinese- if there is any. I'll ask.
Ok- different versions of boards.
If it helps they are using Altium Designer 6.9

This is not an issue of the bosses being reluctant. I'm facing no resistance here, I'm talking directly to the engineer- but he does not speak English and I'm not an engineer. I can say "export in every format you can" and he has- and I've uploaded that. If there is something specific in terms of file suffix, or a screenshot of where to go to get you something useful, an example of some kind, that helps. 

This constant assumption of bad faith is exhausting and counter-productive. The same time spent complaining in this thread could have been spent providing instructions that a layperson (me) could pass on.

RudolphRiedel, 2018-09-15:
>Does anyone want to ask for something without calling the people being asked liars

As this is clearly is addressed to me, please point out where I did that.
That is just your interpretation of what I wrote.
I was throwing around some facts and assumptions based on these facts.
I did not write that Creality is trying to lie to us here or they acting in bad faith.
The renaming of the .PCB to Ender-3.PCB is easily explained by the fact that this is the Ender-3 repository and this was done in order to have everything named alike.
All the files could have chinese names, again I am not saying that would be bad, this is just a possible explanation.

I saw no statement before that Creality is developing this in-house so the assumption that this is a contract work was only made to explain why  after 2 1/2 months the schematic is still missing.
And this would be a totally legit explanation.
I am glad that this assumption is not correct as this means Creality must have the schematics.

>or making accusations of duplicity in the same post?

And I do not get how you come to this conclusion either.
Please point it out to me.
All I see is that I pointed out the fact that the data is not complete without assesment as of which this is the case. And I offered help with this by listing all the extensions of the files that should be in the directory and that are now missing.
The Ender-3 is sold with the V1.1.3 board but the data released is for the V1.1.2 board - it says so in the silkscreen. Again, nothing but a fact.

>If it helps they are using Altium Designer 6.9

Yes, this helps since in that case it really is not a Protel file and I was wrong in assuming it is.
I just checked the Ender-3.PCB with a hex-viewer and found the string "PCB 6.0 Binary File".
While older and a few things have changed since this, this still is a Altium Designer file and at least files saved with AD17 still have that 6.0 tag.
I was wrong and I admit that.

However, you could have pointed out that I am wrong two months ago.
And when I right-click a pcb file and select save-as, only PCB versions 3.0 and 4.0 get named .PCB while 5.0 and current is named .PcbDoc.
Sure, it is totally possible that the engineer uses old file extensions, really old file extensions.
But I should be allowed to find that strange.
And me finding that strange does not mean he is doing it wrong.

Please just let him .zip the directory he is working with.
Maybe make a copy and clean out a few sub-directories before that like "_Previews", "History" and "Project Outputs...".
On August 16 I offered to point out the needed files if you provide a screenshot of the directory.

I am really trying to be helpfull here.
And just look at the timelime and the posts, if that does not reflect patience I do not know what would.

RudyFiero, 2018-09-15:
> "Does anyone want to ask for something without calling the people being asked liars"
> 
> As this is clearly is addressed to me, please point out where I did that.

It was **not** addressed to you. Not specifically I would think. Yes you have been coming across as ... not someone I would want to know. 

> If she isn't getting the kind of material that would be useful then I think it is because whoever she is dealing with is either not being as helpful, or they have been directed to not provide very useful information.

This was my comment from a previous post. And based on the information I have seen, I do wonder if the intent is to **appear** to be open source but to limit the usefulness of the presented material. 

My thought is that if someone wanted to use the design information to copy, and produce a knock-off product, that it would not work. I don't know if that is the intent, but it would be perfectly reasonable to expect that. Even if someone wanted to produce a copied product I doubt if it would be successful. I don't believe that kind of startup could compete with Creality, on price, and considering reputation.

If an individual wanted to use the design to make their own printer I would think it would end up costing more than to just buy the Creality product. The idea that we could save money by doing it yourself often doesn't add up, but I'm sure there are some that think that. 

I think that having the files open source, accurate information, helps enthusiasts to push the design a little further. They are not a threat to a company like Creality, but a source for potential advancement and should be encouraged.

I understand that there are translation issues but I would be surprised that is the real problem. I have not gone so far as to call anyone liars. But I wonder why accurate and current version information has not been released.  Even if it were older versions, but complete and accurate, I would have less doubt.

sexycyborg, 2018-09-16:
Ok nevermind then. It's compliant. That's what you get.

wjsteele, 2018-09-16:
> _"However, you could have pointed out that I am wrong two months ago."_ - @RudolphRiedel 

How could she have done that... she said herself she has no knowledge of the toolchain.  You're so smart, you should have figured that out yourself way before she should have been able to.  It's attitudes like that which are the problem here.

All anyone has to do is ask nicely for what it is they are looking for.  Instead you cop an attitude like you're superior in some way.

And I agree with RudyFiero, you're not someone I'd want to know personally as well with that attitude, however, I have to disagree with him about the rest of his comment.  The assertion that they are just trying to appear to be open source is just a damn stupid thing to say... that's just as accusatory as what RudolphRiedel said.  That doesn't help the situation one bit... instead, just as it happened, it drives them away from trying to cooperate with you.

Jeesh, guys, get a clue here... OSHW is entirely new to the Chinese... they work on a totally different philosophy, and despite what you think... you can't change their ways overnight.  They have no concept of some of the things we do, or even an understanding of why we do them.  Don't go off on me about "Well they should..."... it's not going to happen... especially overnight.  We both need to work together to move forward.

She asked what she could get you to help you out... and you both blew it off and instead continued with your own diatribes about what you want and made accusations based on (false) assumptions.

How about a simple, "Hey, @sexycyborg, how about if we start off asking them for the schematics for the 1.1.3 boards? Those would be the most helpful right now."  That's simple enough and friendly... and then we can work it up to more details later.

DuncanAmos, 2018-09-16:
I agree with Wm Steele, on all points.

To traditional manufacturers, with a traditional business model, Open
Source makes absolutely no sense whatsoever - you put in, and pay for, a
huge number of design hours and then give your designs away for free???

That is a difficult concept to grasp anywhere, and much more so in a
country as traditional as China.

Naomi's article in Make Magazine #61 tells how difficult it has been to get
Open Source even considered in China - maybe those who believe it's easy
should read that...

Yes, we'd like the schematics for the 1.1.3 boards but let's remember to
ask nicely and not demand them...

Naomi is doing the best she can, let's support her and try and understand
the problems she's up against...

*"However, you could have pointed out that I am wrong two months ago."* -
@RudolphRiedel <https://github.com/RudolphRiedel>

How could she have done that... she said herself she has no knowledge of
the toolchain. You're so smart, you should have figured that out yourself
way before she should have been able to. It's attitudes like that which are
the problem here.

All anyone has to do is ask nicely for what it is they are looking for.
Instead you cop an attitude like you're superior in some way.

And I agree with RudyFiero, you're not someone I'd want to know personally
as well with that attitude, however, I have to disagree with him about the
rest of his comment. The assertion that they are just trying to appear to
be open source is just a damn stupid thing to say... that's just as
accusatory as what RudolphRiedel said. That doesn't help the situation one
bit... instead, just as it happened, it drives them away from trying to
cooperate with you.

Jeesh, guys, get a clue here... OSHW is entirely new to the Chinese... they
work on a totally different philosophy, and despite what you think... you
can't change their ways overnight. They have no concept of some of the
things we do, or even an understanding of why we do them. Don't go off on
me about "Well they should..."... it's not going to happen... especially
overnight. We both need to work together to move forward.

She asked what she could get you to help you out... and you both blew it
off and instead continued with your own diatribes about what you want and
made accusations based on (false) assumptions.

How about a simple, "Hey, @sexycyborg <https://github.com/sexycyborg>, how
about if we start off asking them for the schematics for the 1.1.3 boards?
Those would be the most helpful right now." That's simple enough and
friendly... and then we can work it up to more details later.

RudolphRiedel, 2018-09-16:
Stop interpreting plain text to the point of assuming what others think.
All you do is project an agenda upon me that I do  not have.
Read what I wrote, do not try to apply what you think of it to what I might be thinking 

> Instead you cop an attitude like you're superior in some way.

I have no idea where you are getting that from.
All I did was to provide information that has been said is missing.
Anyone working with Altium Designer could have provided that information, anyone.

>All anyone has to do is ask nicely for what it is they are looking for.

Please do.

RudolphRiedel, 2018-09-16:
> On the V1.1.2, the USB pins come from the HC541 buffer but there's no input. 

Hmm? The USB goes thru the FT232RL which is U1 and come out as RXD/TXD on Pins 1 and 5.
The signals are named AIFO (Pin 9 of the controller. RXD0) and AOFI (Pin 10, TXD0).

The SD pins are fully connected as well but most are not named on the SD side.
Only SCLK is named but wrongly so since that is also the name of the input signal.
This results in an airwire but that is all of it, a mere inconveniance.

I have a V1.1.2 board from the CR-10 on my desk and it works.
And it really looks like the data Creality provided.

DuncanAmos, 2018-09-16:
> Just to let you all know the V1.1.2 board is definitely not usable. On the
> V1.1.2, the USB pins come from the HC541 buffer but there's no input. Also
> the SD card pins are not fully defined or connected. So with that said it
> would be kind of Creality to put the Ender 3 V1.1.3 (or V1.1.4 if
> available) PCB files up for us as per OSHW agreement.

As far as I'm aware (and I'm certainly no expert), the current release is
v1.1.3.

Let's not muddy the waters further by asking for something that has yet to
be released for sale.

Gazza-USA, 2018-09-16:
I corrected my comment.  You guys are too quick!!  The v1.1.2 might be OK.  I don’t have schematics to double check.  However I did fix a load of issues on the PCB like badly defined holes and copper voids/no connects.

sl1pkn07, 2018-09-17:
anyone can translate all pcb to kicad format? is more opensource compiant

greetings

adamtegen, 2018-09-17:
If there isn't an automated way, if I had an exported PDF/ screenshot, I
could start working on it in my spare time.

For what it's worth, my guess is it will be somewhat similar to this for
the CR-10
https://github.com/RudolphRiedel/CR-10_wiring/blob/master/CR-10_Wiring.pdf

RudolphRiedel, 2018-09-17:
>anyone can translate all pcb to kicad format?

Have you tried the converter mentioned above?
With the .PcbDoc in AD17 format that I attached on July 6th?

>is more opensource compiant

No, actually it is not.
If Creality would use some chinese program that no-one outside of china has ever heard of and which would be impossible to even get, they would be still in compliance with open source releasing these files.
There is no obligation attached to convert any of it in the flavor-of-the-year file format.

It would be more conveniant, at least to some people.

In this light we can call ourselfs lucky that Creality is using one of the more widespread CAD tools.
And a lot of others release designs in Altium Designer format.
Plus Naomi already tried to supply more variants.
But AD does not really have exporters.

This one for example: https://github.com/ultimachine/Einsy-Rambo

Would you tell them that they are less opensource compliant because they do not use KiCad?

>For what it's worth, my guess is it will be somewhat similar to this for the CR-10

This is a schematic I started from scratch with the available information, 
But not exactly focussed on the controller-board but trying to show how it is connected to the printer.
The only reason I did not release the raw project including CR-10_Wiring.SchDoc so far is that it is still missing way to much.
This is an Altium Designer project as well.

sl1pkn07, 2018-09-17:
> > ¿Alguien puede traducir todo el pcb al formato kicad?
> 
> ¿Has probado el convertidor mencionado anteriormente? 
> Con el formato .PcbDoc en AD17 que adjunté el 6 de julio?
> 
sorry, i'm linux only user

greetings

RudolphRiedel, 2018-09-17:
Try this: https://github.com/thesourcerer8/altium2kicad

[42 more replies]