Creality3DPrinting/Ender-3 #3: [REQ] publish the firmware as a proper git repository

NikiSchlifke, 2018-07-03:
The firmware right now is all in one dump, as it stands there is no way to tell which code came from creality, and which code is from the original authors.
This is also a licence issue currently, as it's not all gplv3. (Not to mention the huge problem a binary blob that contains this code was..)

So if the code already is in version control (which considering the scope of the project seems almost certain) create a new repository on github and add it as a remote.

zoide, 2018-07-04:
I think adding the changes from creality as a PR to Marlin would be a great thing. The hexdump of the resume feature is actually pretty useless in this form. 
Other than that, Marlin 1.1.x already works on the Ender-3.

Nevertheless, cudos for opensourcing all of this.

pyromanci, 2018-07-16:
Currently i'm wrapping up and testing Marlin 1.1.8 with the resume feature i've ported from this code. Should be submitting a pull request to the Marlin bugfix branch in the next couple days along with 2 example configurations for the Ender 3.

stevegibbings, 2018-07-17:
@pyromanci would you post back here with an update on your PR please son can refer to it please?

pyromanci, 2018-07-17:
I am still tinkering with it. It crashed once and failed a second time, but work 3 other times. So the port wasn't as good as it appeared. once i'm done i will commit it to my Marlin Fork and comment here.

rmcolbert, 2018-07-18:
I tracked back the closest commit to the original Marlin repo before Creality forked it. Here's a link to that commit: https://github.com/MarlinFirmware/Marlin/tree/c4fbb44d6d42b84137a89ab7aa5ab7143116184d