Creality3DPrinting/Ender-3 #5: File handling

techmach, 2018-07-11:
Fix the SW files so they are properly exported and save the Solid Assembly file with the sldprt files so you can download all in a single file with all the individual files.  Learn to use Pack and go so they can be properly saved.  Learn how to handle Step files.  These are not script or text files so don't handle them like that sort of files.  Zip them into a single file so you can download everything at once and not piecemeal.

https://www.bing.com/videos/search?q=how+to+export+solidworks+files+to+an+assembly&qpvt=how+to+export+solidworks+files+to+an+assembly&FORM=VDRE

oziee, 2018-07-24:
geez.. your a very rude, abusive, demanding person aren't you!!.. two posts in here, and you think this is how you write things??
you have been on github for 20 days.. if you don't know how it works, go and learn. You don't like the ways its been done, then fork it and make your own.
Learn yourself!

techmach, 2018-07-24:
The SW files are corrupt.  STEP files are not supposed to be handled as TEXT files because they aren't TEXT files.  You can't change the reality because you don't like the perceived tone of the message.

francisalbar, 2018-07-25:
Git repositories are not limited to text files - they work with any files including binaries, images, etc. What you are viewing right now is the Github web interface which attempts to detect mime-types and utilizes an HTML based previewer.

The web interface is not the only way to access these files. If you prefer more granularity or prefer to view these files with your own preferred viewers, please use a different interface such as a GUI Git client like SmartGit, Tortoise, or Github Desktop.

TLDR Learn to Git.

techmach, 2018-07-25:
STEP files like DXF or IGES don't require viewers and they were never meant to be handled like simple ASCII text files.  These file types are generated via translators from 2D and 3D CAD/CAM systems from the 3 dimensional data resident in the source file on that CAD/CAM system.  STEP files are not meant to be edited by hand as are Arduino or Marlin files.   At issue here is that Github handles them as if a user needed to use snippets of code for programming devices.  The clue is the file extension which is everything in this case. The STEP extension has to mean that your system understands that this file is not a man editable file and doesn't need to be read in an editor unless the user wants to.  The end result should be that I can simply right click and save the link as a *.STEP file with the name and the extension intact as I would to with a *.sldprt file that is also hosted here.  

Do you understand this concept?

techmach, 2018-07-25:
My attitude is irrelevant.  What is relevant is performance.  You either can or you can't.  If this is supposed to be a file handling hub it should learn how to handle files.  The Empire will bring order soon enough.  At least Microsoft understands the concept of file extensions and their significance.

techmach, 2018-07-25:
Github's own definition:

"Git

Git is an open source program for tracking changes in text files, and is the core technology that GitHub, the social and user interface, is built on top of."

As I explained earlier STEP files are not just TEXT files and as such are not meant to be handled as TEXT files.  Sounds to me like STEP files just stretch the "Git" to the breaking point.  Better Git with it or be left behind.  There's a whole wide world out here.

TLWhite, 2018-07-25:
techmach, you are correct, the STEP files that are uploaded are corrupted, and are of little use.

Naomi is working on getting us a properly exported "Pack and Go" file for the community to work from.

francisalbar, 2018-07-25:
The Git Version Control Platform handles non text files perfectly fine. The Github WEB INTERFACE is incorrectly treating STEP files as text, and as such, the Github WEB INTERFACE is causing your issue.

If you would like the Github WEB INTERFACE to handle it appropriately, log a defect to Github, or better yet, submit a pull request that fixes the issue. You know, the way open source works. Creality has nothing to do with how the files are handled by the interface.

Your other option is to utilize a different interface to handle binary files directly as I've mentioned before. You won't have to worry about an HTML interface or your browser misrepresenting the files.
- https://git-scm.com/ Command Line
- SmartGit
- SourceTree

9ae8sdf76, 2021-07-19:
@TLWhite @techmach et al,

It sounds like the repository needs a `.gitattributes` file that defines the `.STEP` files as binary files so that a) git won't try to change line endings, and b) won't try to diff them, or c) anything else that happens with text files that would corrupt binary files.

Creating a file at the root of the project called `.gitattributes` and adding the following line would solve this problem:

```
*.step binary
```

Source: https://www.git-scm.com/docs/gitattributes#_using_macro_attributes

PS. @techmach, while you mean well and are technically correct, the way you come across is off-putting and aggressive. This does not encourage others (as you've witnessed) to listen to what you have to say. In fact, it often times will have the reverse effect. It's also worth while to remember that not everybody knows what they don't know, not should they be expected to. We are all here to learn and help, and should approach these kind of situations in that spirit. :)