Creality3DPrinting/Ender-3 #6: no file , bed level ender 3

djawho, 2018-07-11:
hi , i don't have the file "bed level ender 3" in my sd card . And it's not here . Where can i dowload , please

elmerohueso, 2018-07-14:
If it isn't here and wasn't on your SD card (it wasn't on mine, either), where did you hear of that file and what do you expect it to contain?

probonopd, 2018-07-15:
https://www.thingiverse.com/thing:2987803 is very cool, maybe it could be put in the firmware menu so that one doesn't even need an SD card?

pyromanci, 2018-07-16:
there wouldn't be enough room in the firmware for it. The chip only has 128Kbytes of space.

stevegibbings, 2018-07-17:
I assume you mean a gcode script to support manual bed levelling. I use the one on this page 
https://www.nolimits3d.com/pages/downloads

steenerson, 2019-12-13:
Closing out this old issue, the stock SD card contents are in the repo

https://github.com/Creality3DPrinting/Ender-3/tree/master/Ender-3%20SD

closed 2019-12-13 by steenerson