Creality3DPrinting/Ender-3 #8: Firmware test

TalZiv, 2018-07-16:
Hi,

Did anyone actually flashed the firmware posted in the repo the HEX or the actual files?
Or did some one dumped the firmware from the printer itself and compered it to the HEX?

pyromanci, 2018-07-16:
I've not tired the Hex File, but did try the code after I put the bootloader on the board. Which did work for me.

TalZiv, 2018-07-16:
And everything works fine?, If so this is great news, a lot of people wanted to know that the files are OK.
and have the option to go back to original firmware from marlin.

pyromanci, 2018-07-16:
From what i can tell. I didn't test every feature. It compiled flashed and booted, heated the hot end, and moved it around a little.

Though honestly I wouldn't go back to stock. It's based on Marlin 1.0.X. So it's a bit of pain to flash if you flash anything else. Need to use a different version of arduino (i ended up using 1.0.6), libraries from the Marlin 1.0.X branch, reburn the boot loader from that version of arduino (which was odd), ect ect. and vsersa to back to another copy.

rmcolbert, 2018-07-17:
I have done both. I compiled (after fixing issues caused by newer versions of the Arduino Studio) and everything worked. I also flashed the hex file directly and it went fine as well.

pyromanci, 2018-07-17:
Might be worth while to create a pull request with the fix to allow it to compile with the latest copy of arduino.

Gradimilo, 2018-08-09:
@rmcolbert what version of Arduino IDE did you use ? I'm with a problem in the LCD. The graphics are distorted. I had try Arduino IDE to 1.6.8 version, change u8glib version, compare with others marlins versions and nothing has worth.

rmcolbert, 2018-08-14:
I used Arduino 1.6.8 but had to downgrade u8glib for the graphics to work.

Stev-O95, 2018-10-19:
Hi,

i would like to Update my Ender 3 by using the "Ender-3-Power-Fail-Resume-EN.hex" file by using Octoprint running on my raspi. Thanks for the help.

rmcolbert, 2018-10-19:
>           Hi,
> i would like to Update my Ender 3 by using the "Ender-3-Power-Fail-Resume-EN.hex" file by using Octoprint running on my raspi? Thanks for the help.

You will need to install the bootloader to use the .hex file if you haven't already done so.

steenerson, 2019-12-11:
Closing out this old issue, the firmware source+hex have been updated to 2019-04-26 version and compiles on latest (1.8.10) Arduino as-is.

closed 2019-12-11 by steenerson