Creality3DPrinting/Ender-3 #12: Thosiba FashAir not work in the ender3. seems need more power in the microsd slot

sl1pkn07, 2018-08-06:
Hi

I want add the Toshiba FlashAir W-04 16Gb (Wifi module + SD card) 

https://www.flashair-developers.com/en/discover/overview/

 in the ender3 with this microsd/SD adapter

[image]

the SD part is read OK (formated in FAT32) in the printer, but whe wifi module seem need more power signal. because the module is never active due a low power signal from the microsd slot (tested with PC/camera with/o external card read. all work OK. but in the ender 3 the wifi never activate (i can't ping to the unit)

is possible, in the new revision of the board. add support for this type of SD?

how i can add support/hack for this card in the actual 1.1.3 revision of the mainboard?

datasheet
https://www.toshiba-memory.com/products/toshiba-wireless-sd-cards-flashair-w-04/?pdf
http://www.geocities.jp/bokunimowakaru/diy/arduino/flashair.html (japanese, extracted from https://www.flashair-developers.com/en/support/forum/#/discussion/70/power-requirements-for-cards)

greetings

RudolphRiedel, 2018-08-17:
The boards design is generally flawed in this aspect since the SD-Card is powered by the internal LDO of the FT232RL which ihas a nominell output current of 50mA.
This is not even enough for reading from some cards,.
[image]

In order to hack this one would need to cut the highligted trace and connect an additional 3.3V/200mA regulator to it.
I would cut it on the top side right before C31 and connect the 3.3V to C31.

>how i can add support/hack for this card in the actual 1.1.3 revision of the mainboard?

Wait, what? There is a 1.1.3 out?
The picture is from the 1.1.2 data.

sl1pkn07, 2018-08-17:
Hi

yep. my ender comes with the 1.1.3 revision of this board

you can see one here https://www.amazon.com/Creality-Original-Mainboard-Replacement-Control/dp/B07FYJVW1M

RudolphRiedel, 2018-08-17:
Just looking over the pictures and the V1.1.2 data I have no found any differences so far.
There must be some but at least it is safe to say that these are very similar.

sl1pkn07, 2018-08-21:
with regulator, you mean something like this? (stepdown)

https://www.banggood.com/es/Mini-AIO-DC-DC-Step-Down-Module-4_5-24V-Integrated-Adjustable-Fixed-Voltage-Multirotor-Spare-Part-p-1284818.html?rmmds=myorder&cur_warehouse=CN

RudolphRiedel, 2018-08-21:
Basically yes, but I would not recommend this module or any else since I normally do not use pre-assembled modules, I like to design my own boards and I like to be the one beeing responsible in the odd case something goes wrong. :-)

RudyFiero, 2018-08-22:
That power supply in your banggood link is expensive. You could use this instead. https://www.banggood.com/Small-Mini-360-Adjustable-DC-Power-Supply-Module-Mini-Step-Down-Module-p-917568.html?rmmds=category

I have designed that into some of the products at my work. We have used more than 500 so far. It is a good voltage regulator. 

You need to adjust the voltage before you connect it to a circuit. We replace the potentiometer with a fixed resistor so that our customers will not alter the set voltage.

RudolphRiedel, 2018-08-22:
Well, it does say "Input voltage: DC 4.75V-23V" but the Ender-3 has a 24V PSU, doesn't it?
I would go with a 30V rating at least.
And since the cheap modules tend to have high ripple voltages on the output I would test it first outside of the circuit I would like to connect it to.
That second module does look to me like they cheaped out on components, especially with the capacitors.
By testing I mean adding resistors to the output for the max load I expect and checking with an oscilloscope what the voltage looks like.
Would really not suprise me to see ripple in excess of 500mV on the second module.

RudyFiero, 2018-08-22:
@RudolphRiedel  Why would you use 24 volts when there is 12 volts readily available? 

> That second module does look to me like they cheaped out on components, especially with the capacitors.

The capacitors look small but they are 10uF.  Considering that the switching frequency is 340KHz the are adequate. If more filtering is needed then it can be added but for the purpose of powering the FashAir, it would not need more.

> Would really not suprise me to see ripple in excess of 500mV on the second module.

Why would you expect that, do you know how switching regulators work?  The peak to peak ripple is  60mV at 5.2 volts. With the output set for 3.3 volts the peak to peak ripple is 42mV.  That was at 1/2 amp output. That doesn't change much with variations in load. 

The data sheet for that part is here. https://www.mouser.ca/datasheet/2/277/MP2307_r1.9-1384272.pdf

This is a very good synchronous buck converter. The inductor used on the module is very good. The only issue I have is the edge or the board is so close to the connection holes that they is little structural strength. So you need to be a little careful if you use wires connected to it and you are going to flex things.  

I first bought some of these for my own projects. I was so pleased with it that I designed it into a number of products at my work. It was cheaper to buy the modules and populate it as a through hole part than to buy the components and implement the parts as part of the surface mount assemblies. In one design I did need to integrate the converter as part of the surface mount stage. I needed access to the enable pin of the regulator chip. 

The cost of this module is low but it is in no way a "cheap" design as far as performance is concerned. The only thing I have changed is removing the adjustment pot and set the voltage with a fixed resistor. This is so electricians can not adjust it and screw things up.

sl1pkn07, 2018-08-22:
> @RudolphRiedel Why would you use 24 volts when there is 12 volts readily available?

the ender 3 use 24v output power supply

RudolphRiedel, 2018-08-22:
>Why would you expect that, do you know how switching regulators work? 

Because I have seen enough of cheaply made crap already.
And note, not saying that this one is, just saying to be generally cautions with switching regulators, especially inexpensive ones.

RudyFiero, 2018-08-22:
@sl1pkn07  
> the ender 3 use 24v output power supply

I have not received my printer yet. In transit. I saw the silk screen layer marked as 12 volts so I though that was present. If there is not 12 volts then my preference would be to use an available +5 volt supply (assuming 5v is used by the CPU) to generate a 3.3 volt supply. I would prefer not using a higher voltage supply that drives motors to generated a low voltage, low current supply.

sl1pkn07, 2018-08-22:
[image]

RudolphRiedel, 2018-08-22:
And I would not add annother 150,mA+ load to the 5V supply.
The PSU can take it, especially since a switching regulator reduces the 200m@3.3V to about 30mA@24V.
But can the regulator on the controller-board take the additional load?

RudyFiero, 2018-08-22:
It really would help if they released the schematic.

sl1pkn07, 2018-09-06:
Its working!

Pics later

Greetings

sl1pkn07, 2018-09-14:
hi,

bad news. i have notice freezes in the printer like when unplug the SD card durning print

also have notice between C31 and stepdown output have 4.00v+/-, but if unplug the stepdown output wire from the mobo (welded in the C31) , and test the stepdown output without load, have 3,3v

then i back the mobo to start point, and again have notice the C31 to GND is 4.00V!

now in test unweld the pin 4 in the microsd/sd external adapter and connect the stepdown output in it. the voltage is now 3,3v, and the flashair unit work, at te moment, without problem (need test more more time)

greetings

Pointer-de, 2018-12-09:
@sl1pkn07 
Hi. I use the Flashair W-04 16Gb on my CTC Bizer printer, so I want to use it on the Ender3 as well. Can you already say something about your long-term test with the Flashair?

sl1pkn07, 2019-05-26:
sorry. i'm switch to Re-ARM+Ramps some time ago (still work in progress)

greetings

probonopd, 2019-06-30:
A working alternative is https://github.com/probonopd/WirelessPrinting (if you can find a way to connect to RX and TX on the Creality PCB). It uses an ESP8266 or ESP32 for wireless printing. More flexible and much cheaper than a FlashAir.

sl1pkn07, 2019-06-30:
yes, but very very very (y say very?) very slow in file transfer (because the transfer is via serial protocol)

RudolphRiedel, 2019-06-30:
Why would it be painfully slow?
As with Octoprint the serial connection is used to send gcode commands, not to transfer files.

sl1pkn07, 2019-06-30:
becausei don't want any external thing control the printer/prints

probonopd, 2019-06-30:
@sl1pkn07 the software i linked is not slow at all, it streams the gcode to the printer line by line as it is being printed. It does not transfer the file to the printer's SD card.

sl1pkn07, 2019-06-30:
yes, know how works that soft, like the soft i use (ESP3D) and the knowed pronterface. but i dont whant that

i principal use the flashair because i what tharnsfer the files to te print without phisical conenction (the printer is in other room), and the flashair do it flawessy. the ESP-01 unit (with ESP3D) is used only as brigde between pc and the printer only on few cases i need send codes manually

probonopd, 2019-06-30:
Then software I linked works different than ESP3D. It uploads the gcode to the ESP or an SD card attached to the ESP (fast), then you don't need the WLAN connection anymore as it sends the gcode to the printer.

sl1pkn07, 2019-06-30:
both can able send gcode trougth the esp (ESP aso can send codes/control the printer directly from the ESP, wifiprintoing needs slicers (no simplify32 :(). i not need both of that funcionality, only need a simple serial bridge com port over wifi, like the HC-05/06 can do (but over Bluetooth stack). ESP3D include it, wifiprint i'm not sure

siochs, 2019-12-02:
Those FlashAirs are quite expensive and my research lead me to this issue here. 
However, on [this blogpost](https://www.instructables.com/id/3D-Printing-Over-SD-Card-Wi-Fi-Hack/) they wrote that the WiFi feature is only be enabled when you start writing to the FlashAir. But Marlin never writes to the sd card thus leading the WiFi disabled.
A guy named gearsawe commented something I found very promising and may I am be so bold to cite him here:

> So if your printer is running Marlin and you don't feel like messing with editing firmware you may be able to get this to work by using marlins auto run feature. First edit the file on the SD card as instructed above then make a file in the root of the SD Card named auto0.g ,(that is a zero after auto) edit the file and put the following 3 lines.
M28 EnableSD.g
G0 X0
M29 EnableSD.g
this will write one command "G0 X0" to a file named EnableSD.g which would effectively cause the card to enable the wifi.

Could someone please verify if this workaround also works with the Ender3 and the FlashAir? Probably the problem observed here was never related to insufficient power supply?

sl1pkn07, 2019-12-02:
Is problem of the power supply, with the external DC, or with other board, like the ReARM i now use, all works as expected

siochs, 2019-12-02:
I was aiming for a solution without hacking/replacing hardware parts. Something like this may also work in combination with a Raspberry Pi: 
* [sd card multiplexer](https://hackaday.com/2014/06/08/the-in-circuit-sd-card-switch/)
* [sd wire](https://wiki.tizen.org/SDWire)

sl1pkn07, 2019-12-02:
remember, the power supply for sd port in the creality boards comes from the FTDI chip, which can output only 50mA

https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232R.pdf
(pin 17, 3v3out)

RudolphRiedel, 2019-12-02:
And with the new boards they made it worse by switching to a CH340 which does not even have a 3.3 regulator with a current rating but just a reference output.

siochs, 2019-12-03:
Took a look into the pcb files and your fotos. I acknowledge the design flaw. My ender 3 is recognized as CH340 from the kernel. My goodness... My suggestion would be to interrupt Vcc in the SD adapter cable, but I saw you already did that...
> now in test unweld the pin 4 in the microsd/sd external adapter and connect the stepdown output in it. the voltage is now 3,3v, and the flashair unit work, at te moment, without problem (need test more more time)

sl1pkn07, 2019-12-03:
Boards 1.1.4 and 1.1.5 i remember have CH340 instead of FTDI chip

GayCavalier, 2020-07-04:
So here's the lazy-man's solution...    I found a 'plug-in' type DC transformer and set it to 3.3V (supplies up to 2000mA), snipped the end off and wired it directly into the SD>Micro-SD adaptor board.

When viewing connectors facing you, from the left:
Ping 4 is +3.3VDC supply
Ping 6 is GND.

One thing I like about it is that the Toshiba FlashAir drive is still powered and accessible on the network even when the Ender 3 Pro is powered-down.

[image]

RudolphRiedel, 2020-07-05:
At least cut the 3.3V that is going thru the adapter into the board.

siochs, 2020-07-16:
I also experimented with this micro sd adapter (was it from cablecc?) when i tried to switch between the ender and the raspberry using a relay.

Unfortunately, it seems the FlashAirs are discontinuoued. What is left on the market is extremely expensive. 

I was so annoyed by this I built my own WiFi enabled SD card with an ESP32. [See here](https://github.com/siochs/ESP-ShareDisk)

sl1pkn07, 2020-07-16:
Bigtreetech have this https://www.aliexpress.com/i/4000991425856.html

siochs, 2020-07-16:
Dude, you don't know how long I was searching for solutions like those you posted and did not find them. However, I had the parts lying around and learned a lot ;-)

sl1pkn07, 2020-07-16:
that solutions is released from may or june in this year

RudolphRiedel, 2020-07-16:
What is the point of these anyways?
I mean I still need to go the printer and start the print-job from "SD-card".
And at least with the original Ender-3 boards you either need to modify the board or supply these cards externally somehow.

siochs, 2020-07-18:
> What is the point of these anyways?
> I mean I still need to go the printer and start the print-job from "SD-card".
> And at least with the original Ender-3 boards you either need to modify the board or supply these cards externally somehow.

Not with my solution, hit "Autoprint" and toggle your printer power from remote using an intelligent switch. You know Marlin's auto0.g feature do you?

RudolphRiedel, 2020-07-18:
> You know Marlin's auto0.g feature do you?

I did not but I guess this makes this a whole lot more interesting.

siochs, 2020-07-27:
Just saw there is a new version of the Ender-3 on sale (https://www.creality3dshop.eu/collections/ender-series-3d-printer/products/creality3d-upgraded-ender-3-v2-3d-printer). They probably fixed the 3.3V power issue on the SD slot? Didn't find a PCB layout so far...

RudolphRiedel, 2020-07-27:
I found a picture of the 4.2.2 that is used in the Ender3-V2:
[image]

They are apparently using a 2A step-down converter for the 5V, followed by an ANS1117 for the 3.3V.
Oh yes, they went 32 bit, there is a STM32F103 on the board, so the 3.3V for the SD-card *should* no longer come out of the CH340 they are also still using.

That does not mean though that the board works flawlessly with a WLAN card but it is not looking bad either.