Creality3DPrinting/Ender-3 #13: LCD graphics are distorted

Gradimilo, 2018-08-09:
I download the firmware and I use de arduino IDE to upload de files to ender 3, but the LCD screen are distorted and its almost impossible to read the lines. I tried others IDE versions, others u8glib versions and nothing has work.

At others Marlins versions (news) the LCD works fine.

bmcdaniel74, 2018-12-17:
I have the same issue. Has anyone else experienced this?

Snakefreak, 2018-12-20:
I have a Workaround remove the LCD with bezel. Try out. Mount mine with pla printed bolts and washers.

tkuan, 2019-02-22:
I just want to post this solution for anyone has scrambled LCD problem. I got it resolved by commenting out #pragma GCC optimize (3) in ultralcd_st7920_u8glib_rrd.h file.

steenerson, 2019-12-13:
Thanks for posting the fix, I believe this was fixed by an arduino/library/firmware update along the way as it works now without that line commented out. Closing out this old issue.

closed 2019-12-13 by steenerson