Creality3DPrinting/Ender-3 #19: Extrusion Fan doesn't automatically turn off when below 50'C

steckdev, 2018-09-20:
Based on my studying I believe this is caused by the EXTRUDER_0_AUTO_FAN_PIN not being set properly. Please let me know if anyone else has this issue and hopefully we can find a quick resolution to this.

https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/temperature.cpp#L350

Here is where these are set to "-1" and then just ignored in the temperature.cpp
https://github.com/Creality3DPrinting/Ender-3/blob/f903950df9f815ba03f5acf2b0ec9a39e31f2589/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/Configuration_adv.h#L73

Please refer to these files and articles for more information:
https://github.com/MarlinFirmware/Marlin/issues/3234

pusewicz, 2019-05-31:
Yeah, I have the same problem.

buitre-buitaker, 2019-05-31:
The extruder fan is directly connected to the main power rail of the PCB.

steenerson, 2019-12-11:
Hi @steckdev, @buitre-buitaker is correct that the extruder fan is always powered and there is no hardware support for marlin to be able to control it. Closing this issue.

closed 2019-12-11 by steenerson