Creality3DPrinting/Ender-3 #20: Profiles for Cura 3.x and Slic3r

probonopd, 2018-10-04:
Please post profiles for Cura 3.x and Slic3r. 

The profile on the microSD card that comes with the printer is for an outdated version of Cura. 

Trying to [import](https://ultimaker.com/en/resources/21341-old-cura-profiles-in-new-version) it into Cura 3.4 results in

> File Ender 3.ini does not contain any valid profile.

jewirth, 2018-10-05:
> 
> 
> Actually, Cura comes with one now:
> https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_ender3.def.json

Nice! But why has the Extruder Diameter been set to 2.85 mm when using the Ender-3 in Cura?

probonopd, 2018-10-05:
@jewirth possibly this is a bug. I think 2.85 mm is the global default in Cura, so machine profiles that use 1.75 mm need to specify that.

Reference:
https://github.com/Ultimaker/Cura/issues/4468#issuecomment-426884623

probonopd, 2018-10-08:
Some Slic3r profiles are here: https://github.com/sn4k3/Ender3

probonopd, 2018-10-11:
Simplify3D profile by [Nerys](https://www.youtube.com/channel/UCusgqRXg90d7Nau1rqa4WqQ) at https://drive.google.com/drive/folders/0B-XVTPUBrxzwRTVmM0Fwd3hDblE

scruss, 2018-11-10:
If you use Cura 15.04 or CuraEngine with OctoPrint, here's a default profile extracted from `test-dog.gcode` (aka MakerBot's [Niko the Puppy](https://www.thingiverse.com/thing:220609)) on the Ender 3 SD card: [scruss/Ender3Cura15.ini](https://gist.github.com/scruss/63ce52b7057b3802d6b7824c9e788318)

You'll still need to set the printer size (220/220/250), nozzle diameter (0.4) and enable bed heating.

probonopd, 2018-11-10:
Now, how do we convert this to today's Cura? @scruss

scruss, 2018-11-11:
You don't need to: Cura 3.5 includes Ender 3 support directly. If you're meaning CuraEngine, I don't know. only ever used the older one that works with  OctoPrint

joelhuebner, 2019-03-31:
OctoPi uses the 15.0.4 Cura engine. I've downloaded the full 15.0.4 app & built "a profile" that I then imported into OctoPi. It seemed to work, but I'm going to troll the other profiles listed here & see what I can build.  I'll post my results! I'm building a profile for an Ender 3 Pro and a Monoprice Mini v.1 (which I may gut & put an opensource board in it) a piece at a time re-build.

karl0ss, 2019-04-01:
> OctoPi uses the 15.0.4 Cura engine. I've downloaded the full 15.0.4 app & built "a profile" that I then imported into OctoPi. It seemed to work, but I'm going to troll the other profiles listed here [...]

Can you post your Ender 3 Pro profile for the CuraEngine please?

joelhuebner, 2019-04-14:
find the profile above...

steenerson, 2019-12-11:
I have added the latest SD card contents to the repo. Newer versions of Cura support Ender 3 out of the box and older ones can use the profile from the SD card - closing this out.

closed 2019-12-11 by steenerson

ZoomDomain, 2019-12-31:
Here's a video with a link to 8 ender 3 profiles for prusa-slicer https://www.youtube.com/watch?v=wnJ24KTBF7U and he sais there's one on thingiverse too