Creality3DPrinting/Ender-3 #21: DWG files not opening

Nimrodxv230, 2018-10-04:
Trying to view the mechanical dwg files an none of them are opening on my systems. 
they just all come up with the error "drawing file is not valid"

Astra-arch, 2018-11-28:
Same issue. Trying open .dwg files with AutoCAD 2017 - "drawing file is not valid"

kav4ik, 2019-01-29:
Same issue. Trying open .dwg files with AutoCAD 2019 - "drawing file is not valid"

jeroenboumans, 2019-03-06:
Same here with AutoCAD 2018. Same error.

andypl1, 2019-04-13:
Same problem with online browsers. Any success?

Relixtay, 2020-02-10:
Same here with DraftSight and Bricscad. Same error. :-(

DCinAC, 2020-06-11:
Same issue. Trying open .dwg files with AutoCAD 2020 - "drawing file is not valid"

I wonder when will they check our pull request...

Krulk, 2021-05-10:
AutoCAD 2022 "drawing file is not valid" !!!