Creality3DPrinting/Ender-3 #26: Mosfets for hotend and bed

Astra-arch, 2018-12-21:
Hello. There are 3 of mosfets in datasheet named SOT-252 (D522), but I can't find them on the internet. Maybe some replacement? If Somebody got them please give a link or replacement models. Thanks!

Lion5608, 2018-12-21:
Hi, use IRF522, there are plenty of them available.
_Merry Christmas._

RudolphRiedel, 2018-12-21:
These are named REP30N06LE in the PCB file.

Astra-arch, 2018-12-25:
Thanks, but found only R**F**P30N06LE and it's 60V.
On the body of mine, there is an Alfa&Omega Semiconductor logo and D522P marking. 
There is AOD522P and it's 30V N-Channel AlfaMOS in Alfa&Omega Semiconductor's data sheets
Going to replace it with IRF3707ZS
Thank you all for the reply

draugnim, 2019-02-13:
Hi. I have a similar problem. Today my new all metal hotend from AliExpress arrived and I swapped it with the old one. While replacing it I disconnected the old one from the motherboard and when moving the bed or other axes I noticed that the display lights up. After I put everything together there seemed to be no problem. I starded the printer and everything was ok, put when I hit 'Prepare PLA' the nozzle satrted warming up but would exceed the set temperature and throw a max temp error. Sometimes it would start to drop the temperature before reaching the max temperature and throwing a error. 
So, did the induced voltage from the steppers that was generated from me moving them  killed the mosfet while the heater was disconnected? 
I'll post a video of me testing the mosfets with a multimeter when it's done uploading. It seems that all 3 of them are gone, but my bed was working just fine.
@Astra-arch  Did you had any success?

Astra-arch, 2019-02-15:
Hi. In my case it was hotbeds mosfet with burned diode (bed started heating without any control after turning on the printer ). I've just replace broken mosfet and it works fine now. Moving parts can generate voltage that can kill drivers not temp. mosfet, I think.

draugnim, 2019-02-15:
@Astra-arch Thanks for the reply. Yes, I checked all three mosfets with an ESR Meter and all of them are dead. I think that static electricity killed them. So you ended up using IRF3707ZS?

Astra-arch, 2019-02-18:
IRLR3103PBF, 30V 46А [D-PAK]

draugnim, 2019-02-18:
@Astra-arch I have a IRLR3705Z at hand. Do you think it would be an adequate replacement. I would really appreciate if you could take a look.

https://www.tme.eu/gb/Document/17c174c6b2956661b95edec73a0e835d/irlr3705z.pdf

cooldoodsm, 2019-03-12:
@draugnim From my experience with building switching power supplies (a dozen or so, so far--I'm not an expert), I think it would be a problem. 

I'm new to stepper motors, but from what I've researched so far, the _frequency_ has to be varied in order to change their speed (unlike plain old DC motors, which can have their speed adjusted by the _duty cycle_). Assuming the duty cycle stays the same, this will affect the amount of time your FET is able to be on and conducting current without breaching the safe operating area (SOA) limits.

The original FET can handle a lot of current at 25V (rounded up for easiness of calculating). It's relatively linear in terms of derating, at all the different pulse lengths.
[image]

Compared to the above (original FET), the one you suggest has bad current ratings at the longer duration end of the pulse widths, and it's also not very linear. You may breach those limits and end up burning the FET.
[image]

If only the SOA curve was similar, you would have been good to go.

RudolphRiedel, 2019-03-12:
Well, the Hotend and the Bed are resistive heating elements and not steppers or swiching power supplies.

And since these are pulsed with very low frequencies, typically below 10Hz or are even switched on for minutes at start, what really matters for a FET is the RDSon.
The switching losses can be neglected.

If the original FETs really are RFP30N06LE (and we do not know because Creality does not tell), then these have a RDSon of 47mR.
The suggested IRLR3705Z have 8mR.
The bed draws about 8A, probably less.
That makes 3W for the original FET and 0.5W for the IRLR3705Z.

cooldoodsm, 2019-03-13:
Apologies! overlooked the detail about it being the hotend. 

You're right, it's possible that it's drawing pure DC until temperature is reached. According to Google search, the hotend for the Ender 3 is rated 24V, 40W. PSU is 24V, so the hotend will draw at most 1.67A of current (easy enough to handle for many power MOSFETs at continuous DC).

_Most_ FETs aren't meant to be operated at high continuous DC currents at _higher_ voltages, according to their SOA curves. With that being said, I would be cautious and use a FET that can handle continuous DC current. Here's an example of a FET that can handle ~2A continuous at 25V, see page 3 (and really inexpensive too, 30 cents USD): https://www.arrow.com/en/products/tsm090n03cprog/taiwan-semiconductor

draugnim, 2019-03-26:
I replaced the MOSFETs and it is working again. But as it turns out the hotend is also bad. I had to put back the old hotend. The new one has trouble keeping the temperature stable. It constantly shifts +- 5 to 10 degrees Celsius. I don't know if it is the heater or the thermistor. 
in the meantime, the the AOD514 (the original ones) MOSFETs that I ordered online arrived, but I'm gonna keep them as spares for now.

steenerson, 2019-12-11:
@draugnim when you replace the hotend, usually you also need to run PID tuning or else the temperature can overshoot/oscillate like you describe. If PID tuning doesn't fix the oscillations, check that the heater and thermistor screws are tight - bad mechanical connections will increase overshoot.

The bed+hotend mosfets on my 1.1.4 board are also Alpha Omega but marked I4144 / GA2515 which I'm not finding the datasheet anywhere. Wouldn't be surprised if they change it frequently since almost any logic level mosfet with low RDSon that can handle 24V and amps should work fine... The IRLR3705Z ones mentioned above should work fine.

Closing this issue

closed 2019-12-11 by steenerson