Creality3DPrinting/Ender-3 #27: Schematic for the Controller-Board

RudolphRiedel, 2018-12-30:
I gave up on pushing Creality for the Schematic of the controller-board but that does not mean that I stopped caring.
And recent remarks by Naomi on Youtube regarding the Ender-3 still left me a bit angry.

So today I sat down and did my own version of the schematic by building an Ender3.SchLib piece by piece, assigning the footprints used and building it from scratch following the data in the Ender-3.PCB.

https://github.com/RudolphRiedel/CR-10_wiring/blob/master/Ender3_schematic.PDF

Yes, it is not finished yet.
But what has been implemented yet should be accurately reflecting the state the CR-10/Ender-3 board is in - including quite some things I would like to see corrected.

RudolphRiedel, 2018-12-31:
Okay, I just updated that .pdf and now it should be complete.
I also tried to make it more readable.

I am down to this:
[image]

The 10 nets that AD reports to be removing do not show up at all on the board.
There is no bit of copper for any of these ten nets, these do not exist in the board.

One component without a name is removed.
Yet nothing seems to be missing after removing it.
Unfortunately there is no way to display a component without a designator as these should not exist in the first place.
There were more but I found the three unpopulated jumpers at the FETs and renamed them in the board-file to JS1, JS2 and JS3.

The components comments that are changed are for components which have just "Comment" in the comment field and I did not even change all of them.

The change of component parameters are for U8, the micro.
I added two parameters that hold information of what this actually is since "AEMEGA44A" felt so odd.

The added net "Net1k_1" is for the resistor "R1-1" and "LED2" which someone just patched into the board like he was using Paint.

This is a complete Altium Designer project with all the files necessary to make it a project.

Lion5608, 2019-01-01:
Hello and many thanks. It is useful. However, few of us are using Altium designer. We will be stick to use annotated .pdf files. Could you evaluate to use circuit designer or similar in the future?

RudolphRiedel, 2019-01-01:
Okay, for some reason Altium Designer keeps destroying my .pdf files and I tried AD17, AD18 and AD19.
And on top of that I uploaded a file under the wrong name yesterday.

Now this should be fixed, the linke above points to a .pdf that I created with CircuitMaker (after a couple of tries).
Let me know if you find something funny.

And poking around on Github I also found this:
https://github.com/Creality3DPrinting/CR-10
https://github.com/Creality3DPrinting/CR10-Melzi-1.1.2

Very much apreciated these have schematic files.

Makes me wish I had seen this earlier. :-)

Lion5608, 2019-01-02:
Interesting, many thank's. I plan to replace the original board with some better one during the 2019 year. But this is depending on amount of time available :)
I will watch the development closely. If anyone has a good board for ENDER 3 suggestion please post.

RudolphRiedel, 2019-01-02:
An upgrade would be nice, even if it would be only for more program-memory.
But I have not found a board that I really would like to buy.

Maybe we will see some more 32 bit boards in 2019.
I would be in the market for something simple but powerfull, based on the 64-Pin Version of the SAMD51 perhaps.

jgueydan, 2019-01-12:
The Melzi schematic has the motors showing completely different connections than the one you made...trying to wire up a new motor and can't seem to get it working. Thanks for your work on this!

RudolphRiedel, 2019-01-12:
The header J5 clearly shows that the provided "CR-10 Schematic.pdf" is not for the V1.1.2 version of the board but for the earlier revision that did not have a revision printed on the silkscreen.
The V1.1.2 board does not have the jumper to switch 5V from VREG_5V to USB.

This may explain why every component has a different name and almost all headers have their polarity reversed.
In any case, the "CR-10 Schematic.pdf" Creality provided is not the source file, even more so as it is not in binary form but only a picture.

RudolphRiedel, 2019-01-16:
Apart from adding it to CircuitMaker here:
https://workspace.circuitmaker.com/Projects/Details/Rudolph-Riedel-2/Ender3controller-board
I did only release the .pdf so far.

I am still waiting for any response from Creality on this.
But it looks like they left us again after dumping annother bunch of files on Github, that much for "building a community around the repository"...

I give it a few more days then I'll upload the whole project to a new repository.

picman100, 2019-01-16:
Ok , i can defo help to verify it and check the project over, then im going to make some changes and re-up it..

RudolphRiedel, 2019-01-16:
Well, verify, the schematic matches the board file now.
The board file needs fixing though.
And then some. But that would end up in a new board and I do not pick up here that anyone would apreciate the effort.
So for now, my main focus is on repairability.

RudolphRiedel, 2019-02-01:
I put everything here: https://github.com/RudolphRiedel/CR10-Melzi-1.1.2

svenyonson, 2019-04-16:
> Interesting, many thank's. I plan to replace the original board with some better one during the 2019 year. But this is depending on amount of time available :)
> I will watch the development closely. If anyone has a good board for ENDER 3 suggestion please post.

I upgraded my Ender 3 with a Duet 3D WiFi. OMG, what a difference. Gone is the display and that painful interface, and a nice web based UI in its place.  But one HUGE difference is the super quite stepper drivers. Yes, $169 is a lot to spend on a $180 printer, but now I have a very, very nice small footprint printer that is now my main printer. Another reason I upgraded was for dual extruders, which I now have, using an E3D chimera which combines two hot ends into one heat sink. A surprisingly easy upgrade.  

You can find bolt on extruder mounts on AliExpress 
that are exactly designed to mate the chimera to the Ender 3 / CR-10. 

[image]

You can see that I printed an endcap for the left bottom rail and then moved the controller case forward. The Duet controller fits nicely in that box with a few mods (drill/file)

svenyonson, 2019-04-16:
Sorry, github likes to turn pictures upside down

thetazzbot, 2019-12-28:
Ender 3 board uses ATMEGA1284P, your schematic says atmega44..

RudolphRiedel, 2019-12-28:
Yes, I forgot about the old version that I put in https://github.com/RudolphRiedel/CR-10_wiring, thank you.
I just replaced it with a renamed copy of "CR-10 Schematic.pdf" from https://github.com/RudolphRiedel/CR10-Melzi-1.1.2

And yes, I messed up the commit message by writing the file came from https://github.com/RudolphRiedel/Ender-3 ...

funkypawz, 2020-11-16:
Hey, can you please suggest how I can get the text from the LCD screen as an output, which I can then convert to speech via TTS on an arduino (or something)?  I have a group of blind people with whom I work, and I want to make the Ender 3 accessible to them.

wjsteele, 2020-11-16:
It might be better to go to the Marlin github and ask them, as Creality didn't write the base firmware.  But, there is a text file for each language that the code "points" to when it needs to display some text.   My guess is you'd have to add some type of hook to it and then stream that text out to the serial port in order to process it with anything.

It might be better to go a different route and simply use OctoPrint to control the printer, as it's already a web based app and standard Windows/Mac assisted technologies already work with the browsers, so no modifications to the firmware or the electronics would be necessary.