Creality3DPrinting/Ender-3 #28: Fusion 360 file produced

Avaviel, 2019-01-11:
Hi,

I was having trouble opening the mechanical files in Fusion 360. They all needed to be converted by the program before they would work, and that took a few days. However, I now have a Fusion 360 compatible file that could be uploaded. I do not have the authority to upload it myself, so here is a dropbox link.

Thank you,
Paul

https://www.dropbox.com/s/ecwvry3e1042cl0/Ender%203.f3d

steenerson, 2019-12-11:
Hi Paul,

Thank for for providing the file! I have added it here:

https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20Mechanical/F360/Ender%203.f3d

Closing this out

closed 2019-12-11 by steenerson

tomihofmann, 2021-12-06:
Hi!
I'm kinda late to the party, but I believe the f3d file is missing the heat brake/chimney altogether.
Is there a corrected version anywhere?

Thanks!

tomihofmann, 2021-12-06:
Also, in nitpicking mood the labels/names for the 'Part Fan' and 'Cold Section Fan' have been interchanged.

Tom