Creality3DPrinting/Ender-3 #31: Ender3- USB connection randomly getting lost

ender3-user, 2019-02-10:
Hello,
I have an issue with USB connection to my Creality Ender 3.
When printing over Octopi (Raspi) or Repetier/Cura/... (different Windows PCs) garbage symbols suddenly occur in the terminal window, print stops.
I tried different USB cable (long/short/shielded...,, 5V pin isolated or not), result is always the same, failure probabinity remains unchanged.
On the other hand- print from SD worksfine.

So, I hoped to improve the behavior using new Marlin version.
I uploaded the bootloader over SPI from RaspberryPi, which semt to work. 
However, it is not possible to flash the firmware over USB connection of Ardruino compiler, since the conenction is unstable and after several retries will be aborted. The last messages I could see in Ardruino mentioned connection unsync...

So, now my Ender 3  has an activated bootloader with blue LCD and there is no reaction to the control cnob.

What can be done there?
Can I flash the whole Marlin firmware over SPI (I have Raspi or Arduino Nano as possible programmer) or at least restore the original firmware?

Thank you in advance

steenerson, 2019-12-13:
@ender3-user it is possible to flash using an Arduino, here's instructions

steenerson, 2019-12-13:
It's possible to load firmware over serial using an arduino, here's some good instructions:

https://all3dp.com/2/ender-3-with-marlin-how-to-install-marlin-firmware-on-your-ender-3/

Closing out this old issue

closed 2019-12-13 by steenerson