Creality3DPrinting/Ender-3 #32: Compilation troubles

F4HFM, 2019-02-27:
Hello.
I've some troubles trying t "compile" (Verify) the _Marlin.ino_ project:
using Arduino version 1.8.8
using U8glib 1.19.1
declaring "Sanguino" as board, and "ATmega 1284 or ATmega 1284P (16Mhz)" as processor

When I try a "verif" I have a lot of errors:
- multiple declarations of _FAN_PIN_
- _using typedef-name 'fpos_t' after 'struct'_ error in SdBaseFile.h

Any idea of what I missed?
Thank you!

fesiqp, 2019-03-04:
@F4HFM I'm submitting PR to fix this right now.

fesiqp, 2019-03-04:
Fix PR submitted.
PR #33

F4HFM, 2019-03-05:
Hello fesiqp.
Nice, it seems to be fixed!
Thanks a lot, now I'm waiting for receiving my Ender3 Pro to test it :)
Thanks a lot.

tlhintoq, 2019-06-22:
Arduino software 1.8.5 or 1.8.9 makes no difference for this.
Trying to compile Marlin 1.1.9 with Ender3 sample conf files (no changes yet, just the out-of-box files for the Ender3.
Compile (verification) fails saying to choose the Sanguino board.  But that's already the board selected.  Version 1.0.2 as specified int he documentation.
Thoughts?
[image]

ochefe, 2019-06-22:
@tlhintoq can you show me which file is in this screenshot? And which version of the repository are you working with?

Also, can you list me which "Processors" are within the "Processor" dropdown, right below the "Board" one you highlighted

steenerson, 2019-12-11:
I have merged PR 33 and this is complete.

@tlhintoq if you're still seeing an issue let me know and we can reopen this or open a new issue. It's working for me with Arduino 1.8.10 Windows, Sanguino 1.0.3 and U8glib 1.19.1.

PR #33 (2019-12-11): Fix compilation error caused by fpos_t struct
Appearently building Marlin on Arduino IDE version 1.8.x causes
the compilation error `using typedef-name 'fpos_t' after 'struct'`

Renaming `fpos_t` to `filepos_t` fixed the error.
Fixes #32
closed 2019-12-11 by steenerson