Creality3DPrinting/Ender-3 #34: Frequent problem with step motor driver failure

rcoscali, 2019-03-06:
In a short period of time I had two failure of step motor drivers (A4988) circuits. First with Y axis and second with E. I found no other solution than changing the board. Is this a known problem ? Is there something I failed to do to avoid such failures ? Measures with oscilloscope shows no current on the failing driver capacity (c17 for Y) and didn't yet measure it but wwill do shortly, I expect  c25 current to be null.
Can you help about it ?
Thanks

wjsteele, 2019-03-16:
Is your motor current set correctly?  If it is not, it will quickly burn them out.

steenerson, 2019-12-13:
@rcoscali is this ongoing or did you find a resolution? Are we clear to close this old issue?