Creality3DPrinting/Ender-3 #35: What happened?

RudolphRiedel, 2019-03-16:
Here is a transcript of part of the sub-titles from Naomi WUs video on youtube called "Naomi Wu speaking at COSCon'18 (China Open-Source Conference)".
This was back in November 2018:

"I'm an Open Source evangelist."
"I am the person behind the first Open Source Hardware Association certified projects in China."
"Nice words but nothing gets done."
"Everyone talks about Open Source- show me the certification, not just a logo on a PCB."
"As I said, I think that doalog with the community is key to successful Open Source engagement."
"Every file you needed- CAD files, BOM, board schematics, firmware, everything to clone it was posted on GitHub."
"Creality3D has maintained their commitment to Open Source and have continuously released source files and maintained GPL compliance."

"You have to be relatable, not just a faceless company."
"Every successfule Open Source project has an online community."
"If your engineers are not an active participant in those communites, don't communicate with the community, the community will abandon you eventually."
"Just dumping a load of files on Github is not engagement."
"If your engineers can't manage pull requests, can't collaborate with Western developers, you aren't going to reap the full benefits of Open Source."
"Especially with Western developers."
"Then you get nothing."
"All of these things require open-minded thinking that Chinese tech companies might not be accustomed to."

What happened?
There have been no significant contributions to the Ender-3 repository after the first upload.
Almost none of the issues have been resolved.
There is no communication going on with anyone from Creality3D or Naomi Wu on their repositories.
All they do is dump, run and ignore.
And my personal favourite, we still do not have the schematic for the board used in the Ender-3.

wjsteele, 2019-03-16:
But you already did that, didn't you?  "I gave up on pushing Creality for the Schematic of the controller-board but that does not mean that I stopped caring.... So today I sat down and did my own version of the schematic..."

Seems you haven't given up.  Stop worrying about it... you have what you need, obviously, so there is no point in pursuing this further in my opinion.  They've given us 99% of what is wanted... and 100% of what was required (which was only the firmware.)

RudolphRiedel, 2019-03-16:
No, what I did is a reconstruction of the V1.1.2 controller board schematic after the board file they released - and that is most likely not 100% accurate because the board file is not.
That is the board my CR-10 has been delivered with.
The Ender-3 is delivered with the V1.1.3 version of this controller board and we still have no answer to the question what the differences to the V1.1.2 board are.

So, first of all, they did not put out the data for the Ender-3 in this repository. Then the data that they uploaded was incomplete.
That alone makes me wonder why the OSHWA certificate still has not been revoked.
We are talking about 9 months without a real contribution now.

And talking about "what was required".
They admitted by now that the controller board is a clone of the Melzi design which is under GPL.
Their repository is even named "CR10-Melzi-1.1.2".
So they would have been required to release that data when they started selling the CR-10 at least.
The Ender-3 is still in violation of the GPL.
And so is the CR-10 since the schematic that they released with the board data does not match that board data. And it is not even the source-data, it is just a .pdf.
And I pointed that out as an issue almost three months ago.

On top of that you have "Just dumping a load of files on Github is not engagement.".

So, bottom line, we have no engagement whatsoever but GPL violations and a handfull of certificates that would not stand a review.
A little bit of engagement would have gone a long way and changed everything.

And you are telling me this is all fine?
I can not shake the feeling that we have been played and llied to and this makes me angry.

Just look at the issues with this repositories and their contributions.

I am not even making a big fuzz about this, I could post this on twitter and the various facebook groups instead of just here. Plus I could try to contact a couple of Youtubers.
This here will go mostly unnoticed, this is just me blowing off a little steam of disappointment.

RudyFiero, 2019-03-16:
I have to agree with Rudolph. They have not supplied the required information. What they have provided is inaccurate and incomplete. None of the issues have been dealt with. 

> I can not shake the feeling that we have been played and lied to and this makes me angry.

Yes.

bergstar, 2019-03-17:
This issue is close to being trolling by github standards. 

Unfortunately it is true that Creality has proven not to care about open source. And did what all other company in China do - **dump & forget.** And again unfortunately OSHWA dose not care about post open-sourcing support and it is not a requirement. 

And frankly expecting "Naomi Wu" to do anything about it is a bit naive to say the least. After all problem is with a huge companies with thousands of workers - you can't blame person who is not even working there.

So as with any other dead github project - the only way forward is to fork it and maintain it "our self".

sexycyborg, 2019-03-19:
Honestly @RudolphRiedel constant harassment every time I logged was so discouraging it was a large part of the reason I've given up on this repo. Nothing I did was enough- even though no one else had ever successfully done half as much, everyone was always angry and accusing. Why make it personal? How is this my fault? I'm a millionaire 3D printer company owner? I fight for what no one else has gotten before- and I've gotten it, but when I can't I deserve personal abuse? 

I'm only paid to shoot videos for Creality, caring about Open Source does not mean I'm willing to be a punching bag for every entitled rando out there.  I have an enormous amount on my plate, a lot to worry about and precious few resources- why keep helping people that don't give a shit and treat me like garbage? It's just not worth it- Chinese have zero interest in Open Source so I burn bridges here fighting for it, then no matter what I get done I'm still get told it's not enough and am attacked by people like @RudolphRiedel. Why would I keep putting myself through that?

I'm not your dog, I'm tired of people snapping their fingers, telling me to fetch and then smacking me when I do. If you want something from Chinese bosses ask them yourselves, as successful as I've been at times, fighting for your so-called "community" has just left me bruised, broke and exhausted.

RudolphRiedel, 2019-03-19:
This is the way you want to play this, really?
All I see there is an ivite for a personal attack towards you and this is really low and dishonest.
Why do you make this personal about you?
And this while anyone can check out for themself what has been going on here.

So, okay, what did I do wrong?
I provided additional files like the "Ender3_pcb_parts.PDF" that went straight into the repository while I technically have not been asked if this is okay.
I did try to be helpfull and to provide anwers to a couple of questions.
I did offer my help directly here on Github and even tried to contact Naomi in private, but I was ignored.
I defended Creality3D and Naomi a couple of times.
I put quite some hours of my spare time into this without getting paid for it at all or bragging about this on my youtube-channel - because I do care about the product.
I have been attacked and no one stood up to it.
I did not even really ask for aynthing untill three months in and after not getting any answers.
I admitted to beeing wrong when I found out.
I do not like to be ignored, played or lied to.
Maybe I did not use enough smilies.
Oh yes, I expressed my opinion while trying to be clear that this is my opinion and I pointed out facts.

RudyFiero, 2019-03-19:
The problem here was that Creality didn't really buy into the open source mindset. If they did then Naomi would have had the support she needed. Without that support this was bound to end as it did. She gets it, but she is not in charge. She did the best she could.

sexycyborg, 2019-03-20:
Creality will do it if I ask them. I offered to ask them before as you recall and was immediately attacked so walked away. @RudolphRiedel's super creepy demands to my personal email account did not help that situation. I'm still not going to ask them because it's not my job and I don't exert myself for people who attack me for trying to help them.

If something is at the right point in the product lifecycle I might Open Source it, but I'm not going to put a push on. As things stand there's no upside for a Chinese company, attacked if they do, attacked if they don't. It's not really a defensible business decision based on community conduct at this point. Chinese 3D printer companies with no OSHWA certified projects get no abuse or demands because they are not even trying so there's nothing to have a discussion about. Why create that kind of negative buzz around an already leading brand?

Dumping files is- as I've said is a bad strategy and a real problem I've warned other Chinese companies to avoid. I spoke about it at COScom specifically based on my experience with this repo. What I didn't say because I did not want to discourage Chinese companies or give credence to their counterarguments is that the blame is shared. I've pointed Creality engineers at this repo and they saw an absolute dumpster fire they wanted no part of- even though it would save them a lot of work it's just not worth dealing with the crazies. There's a reason the files got dumped and no one is signing up and logging in so you can treat them like coolies.

This behavior is unimaginably stupid and short-sighted. Chinese companies have massive incentives- cash incentives to file dubious patents. Local government is actively hostile to Open Source as it conflicts with targets to achieve a certain number of patents (and so innovation unfortunately in central governments mind- yes I know, there's only so many fights I can fight). The OSH community does not contribute a penny to the work I do here. Despite all the progress, I've made I have no resources and no help with OSHWA work in China. Zero. It's just me. When thermal runaway was burning down houses I took three buses and walked to get to the factory in the middle of nowhere to demonstrate the problem in person. I don't have time or money to take a day off to do that- and I still get screamed at because it's 100% all at once or "They don't really care about Open Source". 

This relentlessly hostile attitude spills over- I scraped up enough money and an invitation letter to go to MWRR this year and I've got attendees making physical threats against me- with not a word from the community. It's insane, you can't keep attacking people who are trying to give you what you want and expect that to work out well.

If we make Open Source happen and you can't be civil and encouraging as things progress it in no way affects China or Chinese people in the slightest- but puts you at risk of being locked out of the hardware ecosystem unless you have the resources to fight bullshit patents on Open Source derivatives in a Chinese court. I just dealt with a Chinese company the other day patenting a modification to a RAMPS board- the community tagged me to fix it for them, didn't give a shit when I got on WeChat and did and that's another burnt bridge with an angry Chinese company that will complain to everyone I'm on the "foreigners side".

TLDR; You got an Open Source Hardware certified commercially successful product from a Chinese company which is far more important in the larger scheme of things than any individual schematic. But short-sighted fuckwits who can't see the big picture of the IP situation in China, be patient and work with what's provided decided to fuck everything up and will now claim we acted in bad faith rather than look to their own conduct.

RudolphRiedel, 2019-03-20:
Stop it with the attacks against me already and apologize.

I sent you three messages and not one was creepy in any way.
2018-09-30 - I reached out thru Facebbok - which you ignored
2018-10-11 - I readched out to your ProtonMail account - which you ignored.
2018-10-27 - My last mail over ProtonMail - which you ignored.

This failure of a repository is not my fault and you obviously have no idea what a dumpster fire even looks like since there was not even much going on here.
We all put on the velvet gloves.
And this now still is not anything remotely like a dumpster fire.

I reached out to you privately to not turn this into into a shit-show.
You chose to ignore me instead of at least answering me that you are not interested in discussing this.
I did not even know that my messages reached you untill you brought it up.

And shame on you to talk about patience when the one thing that has been demonstrated over and over again was patience.
Stop pretending this all happened in the last two weeks.

I can agree to what RudyFiero wrote, at least to a certain degree.
The lack of communication is not entirely on Creality3D though.

bergstar, 2019-03-21:
@RudolphRiedel She is not obligated to reply to you or in anyway care about you or your personal messages. You will not get anyway by being a rude person attacking people on GitHub. I hope you understand that this is an open source project - not a support channel to Creality3D. And even their are not in any shape or form obligated to maintain or support this project here on GitHub.

Please take time to read a few code of conducts:
https://github.com/docker/code-of-conduct/blob/master/code-of-conduct-EN.md
https://github.com/twitter/code-of-conduct/blob/master/code-of-conduct.md

If you are so patient about this project, what stops you from forking and maintaining it your self?

RudolphRiedel, 2019-03-21:
>@RudolphRiedel She is not obligated to reply to you or in anyway care about you or your personal messages. 

Where did I even imply that she is obligated to answer me?
I merely pointed out that she did not answer me, not even to tell me to stop.
She dragged it out while wrongfully accusing me.
And I stopped after three messages roughly two weeks apart to each other, it is not that
I was constantly nagging her.

>You will not get anyway by being a rude person attacking people on GitHub.

Gertring anyway with what now? The only one beeing openly attacked is me.
I could respond much more directly to the accusations but chose not to.
I had left it there but she had to attack me.

I am not aware of any attack towards Naomi here.
If you find something from me that you count as an attack, please point it out to me and I will either try to explain better what I meant then or flat out apologize for it - or both.

>If you are so patient about this project, what stops you from forking and maintaining it your self?

It is not about maintaining, it never was.
Problem still is that the information provided is incomplete and in case of the controller board not accurate while they turned this into a marketing instrument by putting a certificate onto it.
They had to take it one step further.
And on top, they are still in violation of the GPL.
If you can show me that any of this is not a fact but an accusation without grounds I will apologize.

This whole thing is pointless now, the outcome is clear already.
But I will not stand down when I am getting attacked.

bergstar, 2019-03-21:
@RudolphRiedel My bottom line point is - for issues related obligations of the Creality 3D as a company - either related to support or service or legal issue one may contact the company directly using appropriate channels - not github repository. I have no experience in HR - I don't think I can comment any further on that subject. 

When it comes to the OSHWA and GPL - there is ways to complain. For instance:
`Any party detecting an alleged violation can submit a complaint to certification@oshwa.org and OSHWA may, at its discretion, conduct an investigation.`

As and owner of Ender-3 Pro I think it is sad that it ended up being dump & forget repository. At the same time, we have a great starting point to fork the projects and make it really 100% open source.

rex, 2019-03-26:
@RudolphRiedel You are the worst kind of troll. I bet you're a delight at parties. The fact that you can type and re-read what you've said above and not realize that you're not just in the wrong but also a horrible person speaks volumes about where you are at this time in your life.

Who hurt you? What happened to you that you became the kind of person who lashes out at people in an issue in a code repo? Is your life really that empty and sad? 

Go outside. Read a book. Turn the computer off. Your days of being a malicious, petty, neckbearded fucking incel _can be behind you_ if you let them.

@sexycyborg I highly recommend closing and locking this issue, this is absurd that you have to deal with this.

RudolphRiedel, 2019-03-26:
@rex So you are saying that I am a bad person by attacking me?
And you do not have any arguments to demonstrate why I am wrong?
Stop emberassing yourself.

Theaxiom, 2019-04-21:
I agree with @RudolphRiedel while his communication style may be a bit abrasive to those with sensitive skin, I do have to say that the personal attacks against him are absolutely uncalled for and clearly a violation of the community standards. That all being said, can't we all just get along and love each other? I am sure you all have much more enlightening and soul-expanding things you could be doing instead of perpetuating negativity.

closed 2019-12-13 by RudolphRiedel