Creality3DPrinting/Ender-3 #36: Pause does not work instantly.

kaovilai, 2019-03-31:
When pausing print when printing first raft layers, temperature drops as heating elements are turned off but the nozzle still moves around for a little bit as if it;s still printing. Extruder also continue pushing the filaments through the cold head which obviously clogs up the line especially my flexible filaments.

When I pause, it's usually serious issues like leveling is not quite right and the nozzle is literally scratching the board.

It seems to only occur on rafts section that is created by Ultimaker Cura 4.0.0

Pausing are quicker but still not instantaneous on actual models. I assume there is some kind of G-Code buffer that has already been loaded and pause commands aren't executed right away.

Please advise if this is a bug.

bergstar, 2019-03-31:
I have tested pause feature with code from Simplify3D and it is working perfectly. The only issue with pause is inaccurate limit switches which are +/- 1mm or so - which can sometimes results in a layer shift.

kaovilai, 2019-03-31:
> I have tested pause feature with code from Simplify3D and it is working perfectly. The only issue with pause is inaccurate limit switches which are +/- 1mm or so - which can sometimes results in a layer shift.

I have updated the issue.

kaovilai, 2019-03-31:
Seems like commands from the display is sent right away but the module responsible for movements is the issue.