Creality3DPrinting/Ender-3 #37: New Ender-3

Jesscatz, 2019-04-02:
First of, my apologies, this not so much an issue but a report about things I found when I put my second Ender-3 together today.

There is a new spool holder included, it is a little longer and has a special nut that locks like a BNC connector.

The heater block is covered by a black silicone sock.

Much to my surprise the belt pulleys are pressed onto the axis of the stepper-motors now.
So we could not install the vibration dampers I bought in preparation.
 I guess I need to find a replacement for these and figure out how to remove the pulleys without damaging the stepper motors.
These are 2GT with 20 teeth and 5 mm hole?

And when I opened the control-box I found that my Ender-3 was delivered with a V1.1.4 version of the controller board.
[image]

I did not notice it right away but it looks like this new version has a different USB chip.

dodox1, 2019-04-12:
https://www.creality3donline.com/creality3d-new-upgrade-silent-114-mainboard-for-ender-3-pro-ender-5-customized-und-non-standard-matching_p0147.html

Jesscatz, 2019-04-12:
This is interesting and the pictures on there are much better than the quick shot I uploaded, thank you. :-)
So the new USB chip is a CH340G.

Does it say anywhere why it is called "silent"?

At least mine does not seem to be any quieter than the old Ender-3, not that I am complaining, I knew what I was getting when I bought it.
But I still would like to install dampers and could not figure out yet how to remove the pulleys.
I removed the Y-stepper but the pulley refuses to budge.
Okay, maybe I am a little concerned about damaging the stepper.

dodox1, 2019-04-12:
Old stepper motor driver A4988 changed for TMC2208.
This is the reason why is it "silent". Hopefully there is no reason for dampers....

Jesscatz, 2019-04-12:
Watching the video from Creality again I got pointed by Youtube to this one: https://www.youtube.com/watch?v=_wEZOAFZhB4
The pinned comment from Teaching Tech is:
"After reading the comments on here and Facebook, it seems that printers shipped with this board don't have the bootloader or TMC2208s. To get this functionality you need to order this 'upgrade' version from their website. If you just received a printer with the 1.1.4 board, I can imagine how frustrated you are!﻿ "

Well, no, I am not frustrated since I did not expect an upgraded board in the first place.
I would have been a sweet surprise, sure, but I got exactly what I paid for.

Now I still need to figure out how to install the dampers I bought at the same time as the Ender-3.

Well, if the board is the same, maybe one of you guys can figure out how to change the drivers?

Melissa3920, 2019-04-28:
I found this video with a good idea on how to install dampers with press-fit pulleys without a hammer. Yes, I found a video like that too. Anyway: https://www.youtube.com/watch?v=IjzrNoFnbcI&t=602s

Jesscatz, 2019-04-28:
Thank you, that is interesting and I will try this.
I was planning on changing the firmware anyways, now this gives me another reason to do it.

I already bought a "Profi Motor Pinion Gear Puller Remover for RC Helicopter Motor" on EBay,
but it won't be delivered before end of may anyways.

The CH340G worries me a tiny bit.
In the absence of data that supports that this really is working I do not have much trust in that the "Internal 3.3V reference" of the CH340 can reliably power the SD-card.
In case it blows I can have it fixed on short notice, but I rather not have to.
Adding a <5 ct 3.3V LDO like an AP2120N-3.3 would have done the job just fine.

steenerson, 2019-12-13:
@Jesscatz thanks for the report, I'm closing out this old issue.

closed 2019-12-13 by steenerson