Creality3DPrinting/Ender-3 #42: git push failed

schweb5, 2019-10-07:
Hi all,
I'd like to contribute to this repo. I have cloned it and done some simple changes to it.
Now when I push it I get the following error message:

`git push
Fatal: AggregateException encountered.
Username for 'https://github.com': 
remote: Permission to Creality3DPrinting/Ender-3.git denied to .
fatal: unable to access 'https://github.com/Creality3DPrinting/Ender-3.git/': The requested URL returned error: 403`

Can anybody help please?
Thanks in advance

mavenius, 2019-10-07:
It looks like you're trying to push directly to this repo, not your clone of it. What you want to do is push to your clone and then open a pull request into this one. That is a formal request for your changes to be pulled into the repo to which you don't have write permissions. A maintainer of this repo can then review that request and go from there. Hope this helps!

steenerson, 2019-12-11:
Hi @schweb5, I'm closing this issue since as @mavenius mentioned, the proper method to propose changes to this repo is to submit a pull request.

closed 2019-12-11 by steenerson