Creality3DPrinting/Ender-3 #45: PCB and BOM for 1.1.5 Silent Board?

fivesixzero, 2019-11-19:
While working with my new V1.1.5 Silent Board I’ve found some subtle but important dIfferences between the 1.1.2 board detailed in this repo. It’d help a lot, especially with any future repairs, to have an accurate BOM and layout for this board.

The main difference I’ve noticed is the use of R150s for the sense resistor pairs instead of R100s. Based on the TMC2208 data sheet it looks like this effectively caps the output of the motors at about 15% below the max RMS for the 2208 part at 1.28 Amps (against a 1.4 A RMS max in the 220x data sheet).

That seems like a wise decision with the new drivers, helping to avoid burning out the chips. With this design, if Vref is set to 2.5V or higher it would avoid an over-current situation in the driver. Which is good.

Given that minor change I’m also curious what else has changed, if anything, between the 1.1.2, 1.1.4 and 1.1.5 boards as well as what’s different between the 1.1.5 and the 1.1.5 Silent Board beyond the drivers and the sense resistors.

I’m also curious about the stability of the Vref coming out of the trim pots. The 2208 data sheet has some strong warnings about layout and design for stable analog Vref as well as sense resistor grounding. Reviewing the design without reverse engineering the physical board to verify things would require updated design and BOM data.

TL;DR

1. What changed in BOM or layout between the 1.1.2, 1.1.4, and 1.1.5 board revisions?
2. What changed in BOM or layout between the “normal” and “silent” board revisions?

rickyrockrat, 2019-12-28:
They removed the FTDI chip and replaced it with a HL-340. Unfortunately, the USB port STILL powers that LCD backlight when 24V is off. It would be very nice to see a PCB layout so I can fix that like I did on the 1.1.3 version.

RockyTV, 2020-01-07:
They should publish it as RAMPS is open-source, they haven't published the source code for their silent board yet.