Creality3DPrinting/Ender-3 #46: Please respond to the issues and pull requests!

tigert, 2019-12-05:
@Creality3DPrinting and @sexycyborg  - you have a community of people here, and we do value your decision to open up the designs and firmware, but I hope you don't let this effort dry up. There are a lot of issues open for almost a year without response, and even two pull requests which means people are excited and interested and willing to contribute.

3D printing and the whole maker community was born from sharing and brainstorming together. It would be very interesting to see where this could go if we worked on ideas and concepts together.

You made a very bold move to open up - please do not let this die in silence!

I got my Ender 3 because I value your compliance with open source, and because I saw the statement that you wanted to take open source and your community of users seriously.

How can we help you?

sexycyborg, 2019-12-05:
Next time when the maintainer of an Open Source project like this is being subject to significant targeted harassment by the likes of RudolphRiedel the community needs to speak up or understand the project will suffer for their absence. I fought to get the most popular 3D printer in the world OSHWA certified, the first in China and people turned around and demanded my sponsor defund me because I did not work for free on it fast enough for their liking. Just like the community decided that wasn't their problem, dealing with maintaining this repo is no longer my problem. If Creality wants to do it they have access.

steenerson, 2019-12-11:
Hi @tigert, I've been added to this project to help sort through these issues - I hope to get some of the noise reduced in the coming weeks and help direct efforts on the rest.

RudolphRiedel, 2019-12-11:
And again Naomi is attacking me, what a surprise.
Seriously, what is wrong with you?

tigert, 2019-12-22:
Communities consist of people, and that can be both great and horrible at the same time.

I am very happy to see new updates to the repository. All people-issues aside, what @sexycyborg and @Creality3DPrinting have done is important. And thanks @steenerson for your updates!