Creality3DPrinting/Ender-3 #49: Ender 3 making print Taller.

Johnathan2581, 2020-01-07:
Hello,

I'm currently printing with some new PETG I got for Christmas. I started a common (20x20x20 XYZ cube ) print to check and make sure everything is okay. Well, I run into an issue for some reason it has doubled the z of the print and I can't figure out why it's doing so. My sliced CURA file sliced it like a 20x20x20 so I don't think it's that. I also looked at the Z-axis steps and it's set to the default 400/mm. I also took a look at the two little bolts on the Z screw coupler and they are tight. 

Any other ideas would be a great help
Thank you!
[image]
[image]
[image]
[image]

wjsteele, 2020-01-07:
Are you sure you're gcode is not setting the M92 value for the Z axis incorrectly?  The defaults are shown in M502, but what is active is M503.  If not, try setting the steps per mm to half the value, using M92 Z200, which will tell it it only needs 200 steps to make a complete revolution.  This is actually a common failure mode of stepper motor controllers, where they loose a micro stepping increment.  For example, it's no longer at 16th stepping mode... but only at 8th step mode, so it's doubling the distance it is travelling.  The M92 command will fix that... then you can save it using the M500 command... or simply insert it into your start gcode for each job.

Bill

[20200106_165248]<https://user-images.githubusercontent.com/15080496/71859859-50b50880-30a5-11ea-9a61-2da5b39c649d.jpg>
[20200106_165237]<https://user-images.githubusercontent.com/15080496/71859860-50b50880-30a5-11ea-80da-b55bc505bd5b.jpg>
[20200106_165240]<https://user-images.githubusercontent.com/15080496/71859861-50b50880-30a5-11ea-889d-788329a5b7fb.jpg>
[20200106_165244]<https://user-images.githubusercontent.com/15080496/71859862-50b50880-30a5-11ea-8338-f42517ec9935.jpg>

Johnathan2581, 2020-01-07:
Thank you for the advice it was the steps I reduced it to 200mm/s and it printed a pretty good cube
[image]
[image]
[image]
[image]

closed 2020-01-07 by Johnathan2581