Creality3DPrinting/Ender-3 #53: Please fix filename extensions and the whole repo.

i5okie, 2020-05-15:
There are several files where file names overlap.
```shell
$ git clone https://github.com/Creality3DPrinting/Ender-3.git
Cloning into 'Ender-3'...
remote: Enumerating objects: 1273, done.
remote: Total 1273 (delta 0), reused 0 (delta 0), pack-reused 1273
Receiving objects: 100% (1273/1273), 364.64 MiB | 9.28 MiB/s, done.
Resolving deltas: 100% (376/376), done.
Updating files: 100% (1008/1008), done.
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'Ender-3 Mechanical/Ender-3 Parts/M5X8Pan Head Screw.SLDPRT'
  'Ender-3 Mechanical/Ender-3 Parts/M5X8Pan Head Screw.sldprt'
```

Not all systems are configured to be case sensitive. Arguably, most are case insensitive. Meaning that the Solidworks assembly file for Ender-3 is now broken.

jriwanek, 2020-07-16:
Arguably, *windows* is case sensitive, most **other** operating systems can handle mixed cases just fine, I agree with your point however, but inaccurate claims help nobody.

mistic100, 2020-10-10:
I cannot figure out how the repository can have such name conflicts as Solidworks is only available on Windows...