Creality3DPrinting/Ender-3 #55: Creality Slicer Sourcecode

sdaitzman, 2020-07-08:
Hello! Is the Creality Slicer sourcecode available? I would love to refer to it.

cyber-murmel, 2020-07-08:
https://github.com/Ultimaker/Cura

sdaitzman, 2020-07-08:
Yep, I also primarily use Cura, but Creality has their own slicer: https://www.creality.com/download/slicing-software_c0010

I suspect it's based on Cura or Slic3r, but could be wrong. Its rafts are much nicer, and it's only available as an msi Windows installer. I'd love to either port its raft to native Cura, or recompile it for macOS/Linux so I can use it without virtualization.

cyber-murmel, 2020-07-08:
Ah, good to know.

Joe-Bourne, 2020-08-11:
hey @sdaitzman, you're correct - when you load the Creality slicer, look at the bottom right of the splash screen or the Help->About menu and you will see the words "Powered by Cura"