Creality3DPrinting/Ender-3 #58: Ender 3 V2 (4.2.2 Board)

lavanyaarora97, 2020-07-28:
I can't find any board images or diagram. 
Please upload creality

RudolphRiedel, 2020-08-06:
[image]

sagarreddypatil, 2020-12-14:
Is there a schematic for this board?

Ducati87, 2021-02-14:
any news??? is it top secret?

sexycyborg, 2021-02-15:
v2 isn't Open Source

kvartz-tlt, 2021-02-15:
@sexycyborg I have v1 with 4.2.2 board. It is not open source too?

sagarreddypatil, 2021-02-15:
> v2 isn't Open Source

Considering V2 is derived from V1, isn't that a violation of the GPL license?

wjsteele, 2021-02-15:
> isn't that a violation of the GPL license?

No... the author of the original work is free to choose any license they want to.

Jesscatz, 2021-02-15:
> No... the author of the original work is free to choose any license they want to.

Well, Creality is not the author of the original work, it is derived from the Melzi boards which are under GPL.

VinnyCordeiro, 2021-02-15:
GPL is only valid for open source _software_, open source hardware needs an specific license. So, as long as Creality releases the sources for their Marlin implementation, they are not in breach of the GPL.

sagarreddypatil, 2021-02-16:
If the entirety of the Creality Ender 3 v2 is not open source, this would be a clear violation of GPL considering v2 is not using the same code as the original Ender 3(evident by the screen UI), right?

Jesscatz, 2021-02-16:
>GPL is only valid for open source software, open source hardware needs an specific license.

That is an entirely different can of worms.

The point made was that the original author can change the license and while that is true this argument
is invalid for this discussion since Creality is not the original creator of the Melzi boards.

I stongly disagree with the assesment that GPL is an invalid license for open source hardware since we are talking about
the source files here from which the hardware is build, schematic and layout.

But okay let us say, for the sake of the argument, GPL is indeed an invalid license for schematic and layout files.
What does that actually mean?
I means that the files are per default under the copyright of the original author and do not have a license
that can be applied.
It does not mean that the work of the original author is placed in the public domain.
This would make the V1 a copyright violation which is a lot worse than a GPL violation.

This still may not apply to the V2 since it may be that it has seen enough changes to be recognized as original work.

feadot, 2021-05-18:
[image]

RudolphRiedel, 2021-05-20:
That looks helpfull, thanks, I just ordered an Ender3 Max and it should have either this one or the 4.2.7 installed or the next version.

But please provided a source for that file, especially as it is not yours and there is no copyright string on the document and no license information. As it is you are violating someones copyright.

Also, from the looks of it this appears to be redrawn in KiCad and if there is a public repository and if it has KiCad files, this might be even more interesting as this might allow changes and collaboration for these around who use KiCad.

ellensp, 2021-05-21:
Not possible I found it on github, but I can no longer find the original.

RudolphRiedel, 2021-05-21:
That is very unfortunate and I can not find it either.
So we can only hope for now that whoever created the document comes across this.

dwery, 2021-05-25:
It’s probably from https://github.com/Jyers/Marlin/discussions/814

ellensp, 2021-05-25:
yes that is it, Since it been updated I have removed the old one.

apsillers, 2021-07-19:
> But okay let us say, for the sake of the argument, GPL is indeed an invalid license for schematic and layout files.
> What does that actually mean?
> I means that the files are per default under the copyright of the original author and do not have a license
that can be applied.

This is definitely a misreading of the argument about how the GPL applies to circuits.

The argument here, "circuits cannot have the GPL applied to them," is rooted in the basis that circuits are not eligible for copyright, insofar as they are functional articles. Circuit diagrams can be protected by copyright as written/pictorial works (and so the GPL may apply to them, as it may be applied to any copyrightable work) but the physical, functional artifact of a circuit is not copyrightable, so the GPL doesn't have anything to say about it. There cannot be a copyright violation for the manufacture of a circuit for the same reason the GPL cannot apply to a physical circuit: the circuit is not copyrightable in the first place.

From 17 USC S.101:

> Such works shall include works of artistic craftsmanship insofar as their form but not their mechanical or utilitarian aspects are concerned; the design of a useful article, as defined in this section, shall be considered a pictorial, graphic, or sculptural work only if, and only to the extent that, such design incorporates pictorial, graphic, or sculptural features that can be identified separately from, and are capable of existing independently of, the utilitarian aspects of the article.

Related discussion: https://opensource.stackexchange.com/q/2223/50

spuder, 2021-12-18:
@jbrahy Thank you for sharing the schematic. Where did you get it? I believe the UCH245 pinout is slightly wrong. This is the pinout on the 4.2.3 board

[image]

G-Pereira, 2022-07-20:
has anyone found out how to control the part cooling fan?

xcasxcursex, 2022-07-21:
> has anyone found out how to control the part cooling fan?
You're asking in the wrong place and it's very annoying. Use your search engine instead of spamming us all with a notification because you want a howto

G-Pereira, 2022-07-21:
wow lighten up a little sir!

Given this poorly documented board the information on the internet is scarce and that was what led me to post it in here. I am sorry if I annoyed you. You can and should disable notifications if you don't want to be notified about this topic. Actually GitHub has improved its notification system quite considerably and you can fine tune your preferences of what you should be notified by default or not.

Concerning the issue at hand, I could solve it, the fan pin is PA0 which one would easily find by looking at the board file. What I didn't realized is that it is the same pin that drives both the part cooling fan and the motherboard fan. Turns out my part cooling fan was simply stuck on the plastic housing of the hotend, a gentle push solved it.

xcasxcursex, 2022-07-24:
> wow lighten up a little sir!

You realise that you're part of a widespread problem, right? No? Well, now you know. I WAS light with you. You'll know for sure, when I'm not.

> Given this poorly documented board the information on the internet is scarce and that was what led me to post it in here.

Just because you can't find an answer anywhere else doesn't mean post on github in an unrelated issue. You're making the problem worse by decreasing signal to noise, with "how do I use my thing" posts, in a rare place where people are trying to get board files. One more time: YOU'RE MAKING THE PROBLEM WORSE

> I am sorry if I annoyed you.

And yet here you are trying to redirect blame, make excuses, all while doing it again. You're not sorry.

> You can and should disable notifications if you don't want to be notified about this topic.
 
I should, if I didn't, but I do want to be notified about this topic. The topic is sourcing the creality 4.2.2 board drawings. Every single post here except for yours and my replies to them, are on-topic. What you've posted is not on-topic. The problem is not my notification settings, it's you. You can and should post in appropriate places so you don't create annoying notifications for people with the correct settings.
 
>  Actually GitHub has improved its notification system quite considerably and you can fine tune your preferences of what you should be notified by default or not.
  
Actually, no, they don't have an option to filter out off-topic posts. Stop making excuses, you screwed up, just stop.

> Concerning the issue at hand,... Turns out my part cooling fan was simply stuck on the plastic housing of the hotend, a gentle push solved it.

You might want to solve that properly, because what you've described there (improperly mounted hardware prevents fan from turning and cooling) is a potentially serious problem (no cooling + heater = fire), and what you've described is not a solution at all (If a gentle push can unjam it, a gentle push like the G forces of your printer, can jam it). As much of a pain in the back side as you are, I don't want to see your house catch fire. GO ASK SOMEWHERE ELSE HOW TO DO IT. **NOT HERE.**

AkashicSeer, 2023-02-06:
Thanks for the diagrams. 
You got to love these Chinese companies and their lack of understanding of documentation and why such seemingly simple things as standards are important.  They don't comprehend that we the users will flock to the better companies that actually do provide such simple things as documentation, datasheets, specifications etc. etc. etc. 

Why is this such a hard concept for Chinese companies? They stole all of the manufacturing from the West, yet threw out all of our standards and QA etc. Do they really think we came up with all of the standards just to waste time, or does it hurt their ability to be dishonest? Or is it both? In the past you could go to a companies page or open a published book to view specs about components. Now you can't trust a single electronic component, much less complex things like mainboards.

I wish more things were made in places other than China, THEY SUCK AT ELECTRONICS.

xcasxcursex, 2023-02-06:
> Given this poorly documented board the information on the internet is scarce and that was what led me to post it in here.

No, what led you to post in here was a complete lack of regard for the intention of this thread.

> such a hard concept for Chinese

How hard is it to post about the PCB in a thread about the PCB. This isn't the "How do I use my printer" or "let's rant about racist nonsense" thread. Actual tards.

Cozmo77, 2023-04-22:
> I can't find any board images or diagram. Please upload creality
[1623133432-Creality422-Schematic (3)_220129_211224.pdf](https://github.com/Creality3DPrinting/Ender-3/files/11301969/1623133432-Creality422-Schematic.3._220129_211224.pdf)

rennieboudreaux, 2023-08-13:
@ghost Karen, is that you?