Creality3DPrinting/Ender-3 #59: Ender 3 v1 + 4.2.7 v2 Board

badouuuuuu, 2020-08-11:
Hello,

I got a ender 3 v1, with a 4.2.7 board, but i can't found a recent firmware and it's impossible to update to marlin 2.0.6.
Can someone give me more info about this board ?

Flawioo, 2020-08-17:
Hi, the same here. I have dozens of this boards and all of them have no stepper movements. Please can you help all of us?

eduardoluism, 2020-08-20:
Same here, no movements.

christianjou, 2020-09-28:
> Guys, check my issue here, it's fix
> 
> [MarlinFirmware/Marlin#19029 (comment)](https://github.com/MarlinFirmware/Marlin/issues/19029#issuecomment-677572229)
> 
>     * You need to download [bugfix2.0.x marlin files on github ](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip)
> 
>     * Copy Ender 3 Pro 1.5 config in to Marlin folder
> 
>     * Change in to your **configuration.h**  ligne 133
> 
> 
> ```
> #define MOTHERBOARD BOARD_CREALITY_V427
> ```
> 
>     * Change default_env in **platformIO.ini** :
> 
> 
> ```
> default_envs = STM32F103RET6_creality
> ```

thanks a lot. 

However, how activate the Bltouch ? I try but no success