Creality3DPrinting/Ender-3 #60: STL Files?

BenjaminKeating1986, 2020-09-06:
Hi,

I see there are no STL files i have converted all the step files to stl but can not see how i can uplaod these to the repo?

wjsteele, 2020-09-06:
Click the "Add File" in the upper right corner.

BenjaminKeating1986, 2020-09-06:
When i click the upload i get this 

Uploads are disabled.

File uploads require push access to this repository.

zero4281, 2020-09-06:
You could fork the repository and then add them to your fork, and then do a Pull Request.

BenjaminKeating1986, 2020-09-06:
> 
> 
> You could fork the repository and then add them to your fork, and then do a Pull Request.

tried that but seems i can not uplaod a file that is bigger then 25mb :(

zero4281, 2020-09-06:
> > You could fork the repository and then add them to your fork, and then do a Pull Request.
> 
> tried that but seems i can not uplaod a file that is bigger then 25mb :(

For a file that big you might have to do it from your computer instead of the website.  If the command line client seems too daunting, you can get a gui front end for it.

BenjaminKeating1986, 2020-09-06:
ok i think i figured it out, i managed to do a pull request how does that get committed?

zero4281, 2020-09-06:
> ok i think i figured it out, i managed to do a pull request how does that get committed?

Now we wait for the engineers from Creality 3D to review it and merge it in.  Good job!

BenjaminKeating1986, 2020-09-06:
> 
> 
> > ok i think i figured it out, i managed to do a pull request how does that get committed?
> 
> Now we wait for the engineers from Creality 3D, or other people in charge of the repository, to review it and merge it in. Good job!

Great stuff, figured some people may find it helpful to have STL versions, i know allot of people use tinkercad to do quick and dirty designs and you can import your models into that but only supports STL format so mainly for that reason

liialynn, 2020-11-03:
Thanks for uploading the files. But I'm having issues.
None of the STL files are working. what program should I use to open them? I tried meshmixer and freecad but they both fail to import the mesh.

thearchitect2020, 2020-11-25:
Where are the stl files please, do they contain a full 3D model of the ender 3?