Creality3DPrinting/Ender-3 #62: Please provide CAD Model for Ender 3 V2 Mechanical

jottr, 2020-12-03:
Please provide CAD models for the new Ender 3 V2 [mechanical](https://github.com/Creality3DPrinting/Ender-3/tree/master/Ender-3%20Mechanical) components.

nathanshearer, 2021-01-02:
Please also provide them in a format that can be opened by free open source software. Models saved in a proprietary format is not very useful.

One-Armed-T-Rex, 2021-03-30:
You can download the full STEP file here. In inventor you can import is as a complete assembly with all the loose parts

https://d2t1xqejof9utc.cloudfront.net/cads/files/314516fe20262f5973d02158734142fa/original.STEP

lemacs, 2021-04-30:
Does anyone have a new link that works?

Darthkakarot, 2021-09-18:
> You can download the full STEP file here. In inventor you can import is as a complete assembly with all the loose parts
> 
> https://d2t1xqejof9utc.cloudfront.net/cads/files/314516fe20262f5973d02158734142fa/original.STEP

Can we get a link the works pls?

Sukumiku, 2022-05-23:
Here ya go:
https://grabcad.com/library/creality-ender-3-v2-3

Cheers

xcasxcursex, 2022-05-24:
> Here ya go: https://grabcad.com/library/creality-ender-3-v2-3
> 
> Cheers

I can't download from there, the site is full of spyware and won't even load in a secure browser, when I use an insecure browser it asks for an account to download it.

Thanks for trying though!

Panda7556, 2023-10-01:
https://www.printables.com/model/255019-ender-3-v2-step-file-for-easy-modification-and-des