Creality3DPrinting/Ender-3 #63: Ender 3v2 - 4.2.2

konraddg, 2020-12-08:
Please, give me a diagram of the board (from my ender 3v2) 4.2.2. The fans are not running. Diagram needed for repair.

sagarreddypatil, 2020-12-13:
I would also like the schematics to find the Serial RX TX pins to bypass the USB port as I find it too unreliable, but it seems the repo is dead, as none of the new upgrades are on here. So much for the OSHWA certification.

@sexycyborg @Creality3DPrinting

schlameel, 2021-02-18:
Could we please get the CAD (F360) files for the 3v2, including the mainboard?  Thanks.

NickSlash, 2021-02-25:
@sagarreddypatil I've been looking to do that too.

Assuming the CH340G is the USB-Serial interface we need to steal the TX/RX from (its the only one on the board) the image below should show the route from the CH340G to the ARM chip.

Hesitant to test it myself as the vias are a lot smaller than they look in the pictures/videos and my soldering skills are hit and miss.

[image]

acremonezi, 2021-12-28:
> Please, give me a diagram of the board (from my ender 3v2) 4.2.2. The fans are not running. Diagram needed for repair.

Please! 

I am also looking for any diagram.
My ender 3 v2 mainboard stop sending any Z axis output. I am looking for ways to debut and  fix it.

Please anyone can help anyway?

Thanks!!!

xcasxcursex, 2021-12-28:
I have these files. I don't even know where I got them. They are not official schematics and I can't guarantee they are any good at all. But I've used them for a few things, they seem good. Good luck
[Ender.3.V2.4.2.2.-.Schematic.pdf](https://github.com/Creality3DPrinting/Ender-3/files/7785695/Ender.3.V2.4.2.2.-.Schematic.pdf)
[Ender.3V2.4.2.2.-.Schematic.160521.pdf](https://github.com/Creality3DPrinting/Ender-3/files/7785696/Ender.3V2.4.2.2.-.Schematic.160521.pdf)

acremonezi, 2021-12-29:
> I have these files. I don't even know where I got them. They are not official schematics and I can't guarantee they are any good at all. But I've used them for a few things, they seem good. Good [...]

Thank you Very much in did!