Creality3DPrinting/Ender-3 #64: trying to find pin out for creality 4.2.7. retailer claims it is compatible with btt tft35 e3.

cdhunter1010, 2020-12-11:


cdhunter1010, 2020-12-11:
trying to find A pin out for the creality 4.2.7 board. Comgrow claimed it was compatible not seeing any header for it.

jmartens2, 2021-03-29:
Hopefully I’m not too late. could I ask the same question? I’m really interested to find the serial out for a touchscreen board

cdhunter1010, 2021-03-30:
This board doesn't support touch screens. I couldn't get a response from
Creality. I don't think they plan for this to happen with this
particular board. I tested this board for a third party other then the lack
of touch screen support it performed very well.

jmartens2, 2021-03-30:
I love the board that’s why I don’t wanna change to a different manufacturer. But I would also like to get rid of the encoder knob. It’s 2021, for heavens sake! Thank you for the reply. Have a great week!

cdhunter1010, 2021-03-30:
Try octoprint gives wifi ability to the printer and can be customized with
plug-ins to better suit your needs

jmartens2, 2021-03-30:
Info taken. I’m not sure I’m ready for that yet

ellensp, 2021-05-20:
[image]

jmartens2, 2021-05-21:
I’m sorry but I only speak English.

ellensp, 2021-05-21:
The Russian is quite irrelevant, you don't need to it under stand the pin names.