Creality3DPrinting/Ender-3 #67: LCD Screen Not Working

adcallow, 2021-04-28:
BRAND NEW straight out of the box Ender 3. Everything set up properly. Triple checked. Disassembled then reassembled. Followed all the Creality and external Youtube video guides which confirmed everything I've done...still nothing. Best next step?

wjsteele, 2021-04-28:
This isn't a technical support group for the hardware... it's best you go back to the vendor you bought it from for technical support.  This is the public repository for the design files for the printer and no one is here from the company that can help you.

adcallow, 2021-04-28:
Got it, sorry I thought this _was_ the company site, as that is who I bought it from...Thanks for not making me feel like an idiot by incorrectly posting here, I appreciate it...I've seen other boards where I would have been lambasted for such a post. Thx

Gismo12, 2021-08-09:
I recently had the same problem blue screen no response, Using 240v 13 A outlet found a 13A fuse in the mains plug, 15A  fuse on the control board and the capacitor C11 near the X limit switch socket completely destroyed along with the adjacent resistors R11 and R4. replaced the fuse in the mains plug with 2 A fuse, probably try a 5A fuse in the control board. There is also a fuse in the power supply socket Switch which is my next target. When I have finished doing this I might fit another 4.27 board ( fitted with a 5A fuse) and re-flash the the firmware. QC control at the factory might have averted the problem if they had fitted the correct fuse on the control Board.