Creality3DPrinting/Ender-3 #71: Ender 3 Pro BLTouch + Direct Drive + 4.2.7 Motherboard

birdybro, 2021-07-18:
Hi, the latest firmware provided on the website doesn't seem to include the source code for the changes made to Marlin. I was wanting to use your source code as a base of comparison to modify my own spinoff in Marlin 2.0.8. Are there any changes to the source code of Marlin by Creality that aren't the same in their example configurations repo? If so can you provide the source code so I can see it?

Thank you.

starttraveller, 2021-12-11:
I also am having issues with the Ender 3D PRO upgraded with 4.2.7 board and BLTouch. Anyone have Marlin source code which works?

birdybro, 2021-12-12:
I later found it is best to just figure out how to compile the code yourself, unfortunately. Such a huge pain in the neck. Definitely replacing this with a Prusa hopefully soon. Creality's support is lacking, they don't comply with the GPL, and their build quality is getting worse over time.

closed 2021-12-12 by birdybro

starttraveller, 2021-12-13:
Yes, I originally bought the Ender 3PRO which had the 4.2.2 board. I thought upgrading to the 4.2.7 board was going to provide some improvements. Seems that was a waste of $. Then I thought seeing as a lot of my prints weren't sticking to the bed, figured that the BLTouch would have been a good solution. I even ordered the glass bed for ABS prints. Well after so many weeks trying to get the Creality BLTouch to work with the Creality 4.2.7 board and Marlin firmware, I am realizing that in reality it has never been prepared to work together. What a shame! So many unhappy customers. What are they thinking? It would be so easy to make them work together, yet they don't seem interested. Wonder why they are not listening to their customers?

Sent from my iPad,
Regards Lawrence