Creality3DPrinting/Ender-3 #78: Help Needed with Printing

Easylemon24, 2022-01-13:
Hello,

i just bought a Ender 3 V1 and i am pretty new at 3D printing with the Ender. I have the Issue if i try to print even the standard things like the Dog that comes with it starts printing like normal but the Filaments dont stick to the Plate. Does anybody know how i can fix this?I am using PLA

sincerely Easylemon

davemaster, 2022-01-13:
This is a leveling issue. Did You read the manual? Search about "leveling". Also, You can try put some glue to your plate.

Best regards

Easylemon24, 2022-01-14:
Thanks gonna try it out later

Easylemon24, 2022-01-14:
This really worked out thanks again :D

Easylemon24, 2022-01-14:
Okay now i have another Problem, it does print know but the parts dont stick longer then 20 Minutes on the Plate... What can i do?

davemaster, 2022-01-14:
pls, share pictures, sometimes, those says a lot more than words.

Try to work on your slice software, I mean, the slice size

Easylemon24, 2022-01-14:
How do i calibrate it? and i send you some Pictures in a few minutes

davemaster, 2022-01-14:
Did the printer kit come with a Manual-Quick Start guide?

Easylemon24, 2022-01-14:
no

Easylemon24, 2022-01-14:
[image]

Easylemon24, 2022-01-14:
[image]

zero4281, 2022-01-14:
You can find a number of Youtube videos that explain how to level it.

davemaster, 2022-01-14:
First, download or read those files in here:

https://github.com/Creality3DPrinting/Ender-3

let me know when You done.

It's not an easy job to make a calibration, You must learn/read a lot. In Youbute are a lot of material, but, I don't know what exactly 3D Printer do You purchase (without manual? ... a cheap one)

davemaster, 2022-01-14:
try

https://www.youtube.com/watch?v=6PL_rSPZ3M8
or
https://www.youtube.com/watch?v=M8bTi2C67nI

Easylemon24, 2022-01-14:
> It's not an easy job to make a calibration, You must learn/read a lot. In Youbute are a lot of material, but, I don't know what exactly 3D Printer do You purchase (without manual? ... a cheap one)

It was the Ender 3 v1
and i watched the Youtube Video for the Ender 3

I am currently making some tests on my Bed to see if anything is right there...

Easylemon24, 2022-01-15:
Okay i did the "Guide" and it helped very much i found out that idk why i need to move the filament ca. 4 mm away from the extruder and then it prints nearly perfectly.

davemaster, 2022-01-16:
Share a printing picture. Best regards

Easylemon24, 2022-01-16:
> Share a printing picture. Best regards
I just wanted to remove a Print that i did over night and this got left on the plate but i cant get it off...do you know any tips?
[image]

Easylemon24, 2022-01-17:
Okay i got it off i am currently wondering if i can get it to work now because now it does print but the print doesn´t stick and yes i leveled it... if i doesn´t get it fixed with in the next hour i send a pic

Easylemon24, 2022-01-17:
Okay so i got it to work but after the print i am currently doing i am going to clean my Plate thanks to you all for the Help 🗡️

Easylemon24, 2022-01-17:
Okay i thought so until it just got off....

Easylemon24, 2022-01-17:
[image]

Easylemon24, 2022-01-17:
I also dont know if this has something to do with this but the Plate is also very smooth in the middle... i just hope i dont need a new plate/cover...

davemaster, 2022-01-17:
but, the printer is NEW, right? Do an extruder calibration too

Easylemon24, 2022-01-18:
> 
> https://www.youtube.com/watch?v=6PL_rSPZ3M8
> or
> https://www.youtube.com/watch?v=M8bTi2C67nI

Yea i did if you mean this

Easylemon24, 2022-01-18:
Yea i did if you mean this but i do it again today

davemaster, 2022-01-18:
The Z alignment? Then, Print the 20mm cube... you have a Z - axis problem I
think, don't give up
 video(3).MP4
<https://drive.google.com/file/d/1bm-mwoAn_rBVA9fr1rNzEMQ-eS2ltmxp/view?usp=drive_web>

Easylemon24, 2022-01-18:
I cant open the drive link... but i look if i find some tutorial for the z axis

davemaster, 2022-01-18:
this goo... convert into a link instead attach the video,

davemaster, 2022-01-18:
Ill try to share with your email

davemaster, 2022-01-18:
https://drive.google.com/drive/folders/1_vxXh78No0t47qwWoflvh0ZX1Urmu71y?usp=sharing

Easylemon24, 2022-01-21:
I have to say that i did that and I even seaeched for other Videos but it didnt help/ i couldnt do it for e.g. there where Some thing i didnt had on my printer

closed 2022-04-06 by Easylemon24