Creality3DPrinting/Ender-3 #84: ender 3 - temp issue on display with bed and nozzle temp PLZZZZ HELP!!!

mikedattolo, 2023-01-26:
Hey, I have been working on a 3d printer prototype using an ender 3 frame and a btt octopus board and Marlin, my only issue right now with the printer is that when I set the nozzle temp, it sets it and raises it, which is normal, what's not normal is that it also raises the temp number on the bed even though a number isn't set and it doesn't heat either and that when I do go and set a temp for the bed it does heat up, but it is not displaying a temp when heating, it only shows the nozzle temp and doesn't register the bed thermistor and gives an error that the bed isn't heating even though it is, I have tried redefining the pins for the thermistor and changed the pin to another slot, but it gives me a TMC error when doing this, I don't know why or how this is happening either. if I could have help with this, it would be greatly appreciated.

davemaster, 2023-01-26:
Greetings,

First, what firmware are You trying to use...

mikedattolo, 2023-01-26:
> Greetings,
> 
> First, what firmware are You trying to use...

I am using Marlin 2.0.9.5

davemaster, 2023-01-26:
what thermistor in bed? Model and picture pls

mikedattolo, 2023-01-26:
> what thermistor in bed? Model and picture pls

100K ohm NTC 3950FB Thermistor is used, and I should be home shortly to send images

davemaster, 2023-01-26:
As soon you get home, try to check the connection of the therm... with the board... even with a len, check the cables are properly inserted in the thermistor board. This is the most common issue. If the connection are 100% probed well, try with another thermistor, may be you are unlucky and buy a defective therm...

best regards

mikedattolo, 2023-01-26:
I checked both connections and even on the bed as well. Here are some photos I took. The first one is the board with the two thermistors plugged in, the second is what the display normally shows, and then the third photo is what happens when I set the nozzle temp without touching the bed temp at all 
[image]
[image]
[image]

mikedattolo, 2023-01-26:
i hope these help 

> As soon as You get home, try to check the connection of the therm... with the board... even with a len, check the cables are properly inserted in the thermistor board. This is the most common issue. If the connection are 100% probed well, try with another thermistor, may be you are unlucky and buy a defective therm...
> 
> best regards

I hope these help

davemaster, 2023-01-26:
If You remove the bed therm.... what happens? preheat the noozel heat, then disconnect (with the noozle in heat) the bed therm and puth the noozle/head in the bed place... check lectures in the lcd

mikedattolo, 2023-01-26:
> If You remove the bed therm.... what happens? preheat the noozel heat, then disconnect (with the noozle in heat) the bed therm and puth the noozle/head in the bed place... check lectures in the lcd

When the bed pin therm is removed, nothing happens, it still shows that there is a temp on the bed, which obviously isn't normal considering the therm is removed. I tried heating the nozzle, but it still shows the bed heating up alongside the nozzle. When the nozzle is disconnected the bed shows 0 and an error on the nozzle. I tried plugging the bed therm into another port other than the dedicated therm bed port, and it still doesn't change anything

mikedattolo, 2023-01-26:
I also tried switching the pin config with the other port for the bed, and it does show 0, which is new, but.... now when I go to heat it, it comes up instantly with a heating error

davemaster, 2023-01-26:
The thing is, make that change, whatever is working will change temperature, so the voltage over the connection in the board, this is for know if its the sensor/therm or worst, THE BOARD input for sensor, this Analog port can be broken, that's why I would change sensor in the other port to test if as well as it cooled, then show changes in temperature over the LCD

pictures to your code, do You define the EXACT model of board in #define board MY_BOARD (OR Whatever), the  ports (analog) are different between different boards... check it out

davemaster, 2023-01-26:
> I also tried switching the pin config with the other port for the bed, and it does show 0, which is new, but.... now when I go to heat it, it comes up instantly with a heating error

It never CAN SHOWS zero, but the environment/room temperature .... if it is OK

davemaster, 2023-01-26:
the heating error is because once you activated the heating process, the thermistors reading are not showing appropiate lectures, the port of your board can be broken too

mikedattolo, 2023-01-27:
> the heating error is because once you activated the heating process, the thermistors reading are not showing appropiate lectures, the port of your board can be broken too

So I had my friend, who knows more than I look at it, and we actually fixed the problem. Apparently, the bed pin on the btt octopus was broken somehow jumping and connecting to the nozzle thermistor pin which is why it also read the same as the nozzle, and we figured out why we simply couldn't change the pin to another thermistor port and change the config so that it was changed to that new port without an issue and now have resolved the issue and now everything works, thank you for your help it was greatly appreciated :)

davemaster, 2023-01-27:
Great You can fix your issue... best regards

closed 2023-03-24 by mikedattolo