Creality3DPrinting/Ender-3 #85: Ender 3 Filament skipping in extruder motor help plzz

mikedattolo, 2023-01-29:
So I recently just fixed a thermistor issue with a modified ender v2 that I am using for a prototype 3d printer. I am using a btt octopus board and marlin 2.0.9.5 to run this printer, and I just tried running my first print, and it failed because of the fact that the extruder motor is trying to push filament into the hot end, and for a couple of seconds it flows smoothly and then it proceeds to make a noise as the gear pushing the filament skips. However, I have looked at everything by taking apart both the extruder and the hot end, even going as far as replacing the extruder motor, and the issue still is here. Nothing is wrong with either the hot end or the Boden tubing. The hot end gets up to temp, which I have set to 220C, and the Boden tubing isn't clogged or deformed in any way. I have been told to try increasing the temp, which didn't work. I have also ensured that there is no clogging and tried to tighten the spring and handle on the extruder. Yet nothing seems to help my problem. Before I go out and buy a new extruder assembly or a new hot end/nozzle, I was hoping someone could help me figure this out plzzz and thank you

mikedattolo, 2023-01-29:
> So I recently just fixed a thermistor issue with a modified ender v2 that I am using for a prototype 3d printer. I am using a btt octopus board and marlin 2.0.9.5 to run this printer, and I just [...]

Another note to add is that the filament is new but is making a popping sound when coming out of the hot end, and I have tried reducing the print speed to 30mm rather than the normal stock speed at 60mm which didn't work either

mikedattolo, 2023-01-31:
please can someone help

zero4281, 2023-01-31:
I had something similar happen once and all I had to do was take the extruder apart and clean it.

mikedattolo, 2023-01-31:
> I had something similar happen once and all I had to do was take the extruder apart and clean it.

I have taken apart the extruder , cleaned it , checked the Boden tubing , even replaced the motor on the  extruder , yet idk what would still be causing this problem

zero4281, 2023-01-31:
> > I had something similar happen once and all I had to do was take the extruder apart and clean it.
> 
> I have taken apart the extruder , cleaned it , checked the Boden tubing , even replaced the motor on the extruder , yet idk what would still be causing this problem

If you push the filament through manually, does it still have issues?  If not then it's probably related to the extruder, if it is then look closer at the hotend.

mikedattolo, 2023-01-31:
> If you push the filament through manually, does it still have issues? If not then it's probably related to the extruder, if it is then look closer at the hotend.

it flows alright for a couple of seconds but then resists and the filament starts to circle around itself when coming out of the extruder, yet the hotend/nozzle is at 220C so IDK why that would be happening, I disassembled the whole extruder and it wasn't clogged at all either and the boden tube is sitting correctly inside so

zero4281, 2023-01-31:
> it flows alright for a couple of seconds but then resists and the filament starts to circle around itself when coming out of the extruder, yet the hotend/nozzle is at 220C so IDK why that would be [...]

It sound like a clog that you can't see.  I had this happen once.

Try taking the bowden tubing off of the hot end and push the filament into the hotend after it's at temp.  Then turn it off and wait for the hotend to cool down to about 160 and then try pulling it out.  It should have some resistance, but you should be able to get the whole thing out.  This is called a cold pull and it can remove any debris, like carbonized filament from the nozzle.

mikedattolo, 2023-01-31:
I tried this however i could not achieve 

> > > > > I had something similar happen once and all I had to do was take the extruder apart and clean it.
> > > > 
> > > > 
> > > > I have taken apart the extruder , cleaned it , checked the Boden tubing , even replaced the motor on the extruder , yet idk what would still be causing this problem
> > > 
> > > 
> > > If you push the filament through manually, does it still have issues? If not then it's probably related to the extruder, if it does then look closer at the hotend.
> > 
> > 
> > it flows alright for a couple of seconds but then resists and the filament starts to circle around itself when coming out of the extruder, yet the hotend/nozzle is at 220C so IDK why that would be happening, I disassembled the whole extruder and it wasn't clogged at all either and the boden tube is sitting correctly inside so
> 
> It sound like a clog that you can't see. I had this happen once.
> 
> Try taking the bowden tubing off of the hot end and push the filament into the hotend after it's at temp. Then turn it off and wait for the hotend to cool down to about 160 and then try pulling it out. It should have some resistance, but you should be able to get the whole thing out. This is called a cold pull and it can remove any debris, like carbonized filament, from the nozzle.

I tried this however could not achieve a cold pull , something did come out tho , however still having the issues and with the tub removed and manually pushing the filament through I hear a faint clicking noise pushing the filament through the nozzle

zero4281, 2023-02-01:
Okay, so you did remove the bowden tubing from the hotend and tried a cold pull?  Try only heating it to 185 and trying the cold pull again.  You just have to find the right temp to pull at.  Once you get it you should see a larger bit of filament at the end.  If you can't get it I'd try replacing the nozzle, but it could be a small clog.

mikedattolo, 2023-02-01:
> Okay, so you did remove the bowden tubing from the hotend and tried a cold pull? If you're not using the extruder and still have bad results then it's probably not the extruder. Try heating it to 185 and trying the cold pull again. You just have to find the right temp to pull at. Once you get it you should see a larger bit of filament at the end. If you can't get it I'd try replacing the nozzle, but it could be a small clog.

I've tried it 3 times now and have been doing 185C so

mikedattolo, 2023-02-01:
and i tried switching the tip on the nozzle and its still happening and its not a heating issue so

zero4281, 2023-02-01:
Maybe try the filament in a different printer.

mikedattolo, 2023-02-01:
its a brand nre 

> > and i tried switching the tip on the nozzle and its still happening and its not a heating issue so
> 
> Did you try a different filament. Popping can be moisture content.

its a brand new spool of filament got it not even 2 days ago

zero4281, 2023-02-01:
> its a brand nre
> 
> > > and i tried switching the tip on the nozzle and its still happening and its not a heating issue so
> > 
> > 
> > Did you try a different filament. Popping can be moisture content.
> 
> its a brand new spool of filament got it not even 2 days ago

Unfortunately that doesn't answer my question.....

mikedattolo, 2023-02-01:
I dont have any other brands of filament at the moment

zero4281, 2023-02-01:
> I dont have any other brands of filament at the moment

You could try drying over night it in your oven at the lowest temp it will go to, but this isn't a great method.  I have had some success with it though.  Crack the oven door to let the humidity escape.....I'm running out of tips an tricks.

closed 2023-03-24 by mikedattolo