Creality3DPrinting/Ender-3 #86: Nozzle temp keeps dropping and Im not sure why Plllz help!!

mikedattolo, 2023-02-04:
So I am working with a Btt octopus board inside of an Ender 3 V2, and I have been coming here a lot for help because of unforeseen problems. As I keep fixing these issues, I keep running into more. My most recent issue is that I am trying to do a first print, and my nozzle temp is set to 220C, and the bed is set to 50C. It is fine as it starts, but as soon as the part cooling fan kicks on, the temperature crashes down to 200C or lower. It tries to bring itself back up but usually ends with a Thermal runaway error. Yet this isn't the first appearance of this issue, even sometimes without the bed heated or any fans on at all, and I try to heat the nozzle, this also happens. I have a silicone sock on the nozzle, as I was told that it might be losing heat due to a fan, yet this doesn't seem to be causing the issue because I disabled the fans and tried heating it, and it still occurs. I have checked all connections, and there are no loose cables or anything. I have even tried doing PID autotuning with the fans on, as I have been told that this might help, and I still have this issue. I was even told that it could be a bad nozzle and that I should replace it, yet I wanna hear someone else's input. I have even been told that it might be a weak power supply or an electrical issue, yet when asked how to fix it or what is causing it, I was never given an answer. So I would really appreciate some help here, and thank you in advance.

mikedattolo, 2023-02-05:
anyone plzzz help

davemaster, 2023-02-06:
check firmware, the top temperature for your nozzle, it can be

mikedattolo, 2023-02-06:
> check firmware, the top temperature for your nozzle, it can be

It can be what?? and my top temp is 260

davemaster, 2023-02-06:
the MOSFET of nozzle can be then... 
check the nozzle alone

mikedattolo, 2023-02-06:
> the MOSFET of nozzle can be then... check the nozzle alone

wdym can you give a full explanation plz

mikedattolo, 2023-02-06:
> the MOSFET of nozzle can be then... check the nozzle alone

if it were the mosfet it would be a maximum temp error, I am having thermal runaway because of the temp dropping low

mikedattolo, 2023-02-06:
but I have no clue anymore and I need a better explanation

davemaster, 2023-02-06:
> but I have no clue anymore and I need a better explanation

Do You know what a MOSFET is??? NO??? try same nozzle with another BOARD, try also the nozzle alone